OrcaSlicer/OrcaSlicer #7343: Starts and immedialtely crash (Linux version)

Zebulon67fr, 2024-11-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 Linux ubuntu appmimage

### Operating System (OS)

Linux

### OS Version

 Linux Mint 22 Wilma base: Ubuntu 24.04 noble

### Additional system information

Machine:
  Type: Laptop System: ASUSTeK product: ROG Strix G513RC_G513RC v: 1.0 serial: <superuser required>
  Mobo: ASUSTeK model: G513RC v: 1.0 serial: <superuser required> uuid: <superuser required>
    UEFI: American Megatrends LLC. v: G513RC.327 date: 02/16/2023
CPU:
  Info: 8-core model: AMD Ryzen 7 6800H with Radeon Graphics bits: 64 type: MT MCP smt: enabled
    arch: Zen 3+ rev: 1 cache: L1: 512 KiB L2: 4 MiB L3: 16 MiB
  Speed (MHz): avg: 1905 high: 4566 min/max: 400/4785 cores: 1: 4566 2: 4565 3: 400 4: 400
    5: 2283 6: 2118 7: 2029 8: 400 9: 2285 10: 2248 11: 400 12: 400 13: 2030 14: 400 15: 2029
    16: 3932 bogomips: 102211
NVIDIA GA107M [GeForce RTX 3050 Mobile

### Printer

proforge 4 makertech and snapmaker J1

### How to reproduce

Launch the programme (appimage)
Orcalslicer opens and shows all my recent project.
Then stops (closes) without any message.

### Actual results

no message. Need to lauch again and same result

### Expected results

was working before  so should keep working

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17607655/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

I have no clue . It worked perfect until I decide to connect my printer proforge 4 via my network. Test was OK in orcaslicer whit its IP. I can reach the pronter without any issue using its IP in a borwser. But Since orcaslicer refuses to work.

dkowis, 2024-11-02:
I have the same problem right now, I have a pair of Voron printers. 

Here's my log file. I see a webkit bug with a timer, but I'm not sure that's the actual problem.
[crash-popOS-02Oct2024.zip](https://github.com/user-attachments/files/17607846/crash-popOS-02Oct2024.zip)

dkowis, 2024-11-02:
Running it with strace:
```
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
rt_sigaction(SIGSEGV, {sa_handler=SIG_DFL, sa_mask=~[RTMIN RT_1], sa_flags=SA_RESTORER, sa_restorer=0x758dd2a42520}, NULL, 8) = 0
rt_sigreturn({mask=[]})                 = 108625863308720
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
+++ killed by SIGSEGV (core dumped) +++
fish: Job 1, 'strace build/package/bin/orca-s…' terminated by signal SIGSEGV (Address boundary error)
```

alex999888777, 2024-11-02:
https://github.com/SoftFever/OrcaSlicer/issues/7210#issuecomment-2440350811

dkowis, 2024-11-02:
> [#7210 (comment)](https://github.com/SoftFever/OrcaSlicer/issues/7210#issuecomment-2440350811)

Thank you so much @alex999888777 :D Setting the ip:port in the JSON files instead of just the direct IP solved the issue entirely!

Zebulon67fr, 2024-11-02:
Now this is a message when I lauch Orca
[image]

Zebulon67fr, 2024-11-04:
Thank you. This is now solved

rynoian, 2024-11-11:
[image]
freezes at the setup wizard

schenlap, 2025-01-19:
I had the same issue on debian. I start it now with
`env WEBKIT_DISABLE_COMPOSITING_MODE=1 ./OrcaSlicer_Linux_V2.2.0.appimage`
and it works for me.

closed 2025-04-28 by github-actions[bot]