OrcaSlicer/OrcaSlicer #7351: Second Printer Not Selectable

AndOneMore, 2024-11-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10 Pro

### Printer

Creality Ender 3 S1 Pro

### How to reproduce

My Primary printer is a Qidi Q1 Pro and up through version 2.1.0 I have been able to switch between printers from the little drop down menu. However, since 2.2.0 only the Qidi Q1 Pro is visible.  I have gone into the printer management screen, and verified that I have both the Qidi and the Creality printers selected. I am no longer able to use Orca to print with my Ender, please assist

### Actual results

Despite Printers selected in the screen, unable to select other printer from drop down box.

### Expected results

I should be able to see the Creality printer in the drop down menu where it used to be.

### Project file & Debug log uploads

[crash_Tue_Oct_29_16_11_08_0.zip](https://github.com/user-attachments/files/17609149/crash_Tue_Oct_29_16_11_08_0.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanman1, 2024-11-12:
I am seeing the same.  2.2.0 RC works as expected, 2.2.0 Release does not show additional printers even though they are checked.

AndOneMore, 2024-11-12:
I did a little digging. Went into the files and the Creality folder is not being populated with .json files. The folder is completely empty. For the time being, I have switched back to 2.1.0 and everything is working the way it should

closed 2025-02-18 by github-actions[bot]