OrcaSlicer/OrcaSlicer #7353: If I start Orca slicer, it loads all the printers, and if I start it again, it loads only a part of them.

PitFx, 2024-11-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

10

### Additional system information

i5 12Gen 16GB RAM 

### Printer

all

### How to reproduce

Only run Srca Slicer...
If I start Orca slicer, it loads all the printers, and if I start it again, it loads only a part of them. And it's still on every second run around.
[image]
[image]

### Actual results

Only run Srca Slicer...
If I start Orca slicer, it loads all the printers, and if I start it again, it loads only a part of them. And it's still on every second run around.

### Expected results

all profile all time

### Project file & Debug log uploads

[debug_Sun_Nov_03_12_01_26_21876.log](https://github.com/user-attachments/files/17610123/debug_Sun_Nov_03_12_01_26_21876.log)
[debug_Sun_Nov_03_12_02_30_20516.log](https://github.com/user-attachments/files/17610125/debug_Sun_Nov_03_12_02_30_20516.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

igiannakas, 2024-11-04:
@SoftFever I've had a very similar issue with a fresh install of Orca on Windows 11. It wouldn't download my profiles from the Bambu cloud, so I had to copy them manually over to get the printer's setup.

Maybe something is up with the network plugin?  The only change I see related to this is https://github.com/SoftFever/OrcaSlicer/issues/1027 but it shouldn't cause this issue?

FredWang1999, 2024-11-04:
I'm having exactily the same wired issue. After purchasing a Bambu Lab A1 mini and connecting it to Orca slicer (during which I registgered the Bambu cloud account and logged in), my original custom printer (N3DPV2) will only load every other time when the slicer is launched, while the A1 mini is loaded every time.  The following error, along with all the custom filament/printing profile popped out when my printer failed to load:
[error]	2024-11-04 22:48:18.186894[Thread 0x00005c38]:can not find parent for config C:\Users\MyUserName\AppData\Roaming\OrcaSlicer\user\54647047\machine\N3DPV2.json

FredWang1999, 2024-11-08:
> Try copying your config files over to the user profile if you have them available. It fixed it for me.

I tried to copy my configs around with some success. Here's a list of things I did:
Since I wasn't using bambu account prior, there's a _default_ profile folder and a profile folder with my ID _54647047_. The _54647047_ folder contains the latest config for my custom machine and config in _default_ folder is somewhat outdated.

I deleted all the config in the _default_ folder, Orca runs normally but only loads my custom profile every other time

I deleted all the config in the _54647047_ folder , Orca still loads my custom profile every other time, and somehow restored the deleted folder on the firstg run.

I uninstalled Orca slicer and wiped all the configuration folder, then re-installed it. During the setup I first get my A1 mini ready and logged into bambu. Now my custom printer profile stopped showing up. After which I moved the backup of my profile back to the _54647047_ folder, now Orca runs normally but still, only loads my custom profile every other time

For my last try, I uninstalled and wiped orca again. This time instead of setting up my A1 mini, I get my custom printer working first by selecting a random printer in the setup , then moving the backup to the _default_ folder. Now Orca loads my custom profile every time. I then proceeded to set up my A1 mini (downloading network plugin, logging-in to bambu, etc.), after which Orca stopped loading my custom profile, only A1 mini is showing up. Upon checking my profile folder, now _54647047_ is created. Then I proceeded to move my config from the _default_ to _54647047_. FINALLY, now all my machines are showing up correctly EVERY TIME.

So maybe the solution is setting up a random printer profile, copying the custom profile to _default_, login to bambu, then moving the custom profile from _default_ to the new user? Not really sure but at least it worked once for me.

spitfire, 2024-11-13:
I have a similar problem I guess. When I'm trying to sign in to bambu account to sync my Orca settings to a new Mac I'm getting this right after signing in.
[image]

I can't click the cancel button, and regardless of whether I click yes or no Orca just hangs and I need to kill it.

andrempo, 2024-12-18:
I always need to open Orca Slicer twice for my imported profiles to appear.

closed 2025-03-27 by github-actions[bot]