OrcaSlicer/OrcaSlicer #7354: Unable to remove filaments

Coffe83, 2024-11-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

macOS Sonoma 14.6.1

### Printer

Flashforge Adventurer 5M Pro

### How to reproduce

Open a saved Orca project file that has more than one filament. Perhaps even one that has a filament profile that you don't have (I don't know if this is what caused the issue). The attached project file that I opened has filament profiles that I no longer have. Close the project without saving and discard everything. Open Orca again empty. Now the filaments from the previous project is still selected. I found this to be an odd behavior, but this may not necessarily be a bug. But now I can't remove them. The plus and minus buttons that used to be there are no longer there.

### Actual results

No button to remove filaments.
[image]

### Expected results

I should be able to remove and add filaments.
I'm not sure about the behavior that it lists the filaments from last opened project. But this might be the intended behavior.

### Project file & Debug log uploads

[7-Segment Module.3mf.zip](https://github.com/user-attachments/files/17610343/7-Segment.Module.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Coffe83, 2024-11-03:
I now found the issue...
When I opened my printer settings, the "Single Extruder Multi Material"-checkbox was unchecked for whatever reason.
So the real bug here is actually that Orca apparently has changed my printer profile without my permission when upgrading to 2.2.0. In the system presets it can do whatever it want, but if it wants to change my user profiles I like to be informed.

closed 2025-02-09 by github-actions[bot]

darkzeroat, 2025-05-10:
> I now found the issue... When I opened my printer settings, the "Single Extruder Multi Material"-checkbox was unchecked for whatever reason. So the real bug here is actually that Orca apparently has [...]

thats stupid cause im using a sovol sv04 IDEX and i still want to remove filaments that are more than i can use
orca feels so much worse over time, basically nothing is user definable and only what the devs think is important