OrcaSlicer/OrcaSlicer #7356: Allow Orca to register and use its own version of bambusource.dll

dmorris68, 2024-11-03:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I posted this on [issue 811](https://github.com/SoftFever/OrcaSlicer/issues/811) but it had already been closed. It's not technically a bug but more of a feature request, so reposting here as such.

I use both Orca and Bambu Studio slicers for different situations. Orca has some more advanced features, while Bambu sometimes gets a new feature first. Also Maker World requires 3MFs from Bambu Studio. And once I downloaded a 3MF from MW that wouldn't print correctly in Orca, for some reason I don't recall. So I really need to use both slicers.

They each have their own copy of BambuSource.dll -- binary identical, and same CLSID -- in their respective plugins folders that they attempt to load and register. I can choose No to the registration prompt on the "Using a BambuSource from a different install" dialog and the camera still works fine in both. Because they're the same DLL with the same CLSID, only one can be registered at a time. If I respond Yes to let Orca register the DLL, then BS will start throwing the same error. And vice-versa.

Since both are binary identical, I thought I might delete the one that's not registered, assuming the app will find the registered one. Unfortunately, it still sees that it's missing from its own plugins folder and re-downloads it, putting us right back to where we started.

It's not a showstopper, and not necessarily a bug, more a UX issue. It's an annoying popup that I have to dismiss every time I use whichever slicer didn't register the DLL.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I've looked through the code and found the offending section starting at src/slic3r/GUI/wxMediaCtrl2.cpp, in the wxMediaCtrl2::Load method. At around line 136 it checks the path of the registered DLL against its own plugins path, and if they don't match, it throws the error dialog.

I can see at least a couple of ways to address this in the code. One might be to use a unique CLSID for each slicer's DLL, then both could be registered and used independently. Another would be to save the user's response to the prompt and suppress this error in the future, allowing it to continue using the other's registered DLL (which works fine). I kind of like the first option as it allows each slicer to use its own version of the DLL, just in case that was needed one day (perhaps to avoid a buggy release or enable a customization).

I would look at making the change myself and submitting a PR, but it's been ages since I've had a C++ compiler installed with Boost and WxWidgets and the other dependencies I expect I'll need. I guess if the maintainers don't care to try, I may give it a shot. But thought I'd ask.

Noisyfox, 2024-11-04:
> One might be to use a unique CLSID for each slicer's DLL, then both could be registered and used independently.

I'm not sure you could do that as when you register the DLL via regsvr32 the CLSID comes from the DLL itself, and since the DLL is closed source we have no way of changing and distributing that without breaking license or what.

dmorris68, 2024-11-06:
> I'm not sure you could do that as when you register the DLL via regsvr32 the CLSID comes from the DLL itself, and since the DLL is closed source we have no way of changing and distributing that [...]

Good point, I didn't realize the DLL was closed source.

So that leaves the less flexible but still workable approach to suppress the prompt after the user initially dismisses it. Or even simpler, don't even try to validate the path -- just default to using whichever DLL by that CLSID is already registered.

Logic being, since Orca is a Bambu fork and intends to work with their closed source DLL anyway, why not assume it's safe to do so? Yes, there is a risk that Bambu could release a breaking change to the DLL but I feel like the Orca team is tracking pre-release Bambu code and would be prepared to react to it quickly. Seems like the risk would be low.

dmorris68, 2024-12-18:
Just FYI, I'm not sure what changed but I haven't been getting the DLL registration nag when switching back and forth between Bambu Studio and Orca for some time now. Not sure when it happened but it just dawned on me recently that I hadn't seen it in a while. I haven't bothered comparing the latest source, I'm just happy it seems to be resolved.

closed 2024-12-18 by dmorris68

nickchernandez90, 2026-08-20:
I encountered a related registration loop on OrcaSlicer 2.4.2 under Windows 11 with a Bambu Lab H2D.

Orca repeatedly reported that registering `BambuSource.dll` was successful, but live view never loaded. The debug log showed:

```text
wxMediaCtrl2::Load: clsid  path ...\OrcaSlicer\plugins\BambuSource.dll
MediaPlayCtrl::SetStatus: The player is not loaded ... [2:101]
```

The networking plugin loaded successfully and provided a valid RTSP URL. `BambuSource.dll` was also correctly registered under CLSID `{233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA}`. However, this separate registry value was missing:

```text
HKCR\bambu
  Source Filter = {233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA}
```

Orca’s `wxMediaCtrl2::Load()` treats the missing value as error 101. The built-in re-registration flow displayed success for `regsvr32`, but apparently failed to import or persist this separate registry mapping.

Running the following from an elevated PowerShell window, then restarting OrcaSlicer, immediately fixed live view:

```powershell
reg add "HKCR\bambu" /v "Source Filter" /t REG_SZ /d "{233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA}" /f
```

It may be worth changing the registration flow to create this value directly and verify it afterward, rather than reporting success based only on `regsvr32`.