OrcaSlicer/OrcaSlicer #7358: Orca Slicer 2.2.0 won't install on Windows 11

ScrapyCocos, 2024-11-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

11 Pro

### Additional system information

CPU:AMD Ryzen 5 7600X 6-Core Processor
Memory: 32Gb
GPU: NVIDIA GeForce RTX 4070 Ti Super

### Printer

Bambu Lab X1C, Bambu Lab P1S

### How to reproduce

I downloaded the windows installer from Orca Github , double clicked on it.
Run as admin

### Actual results

It doesn't do anything. Not lunching the installer at all.

### Expected results

Starting the installer and install Orca Slicer 2.2.0 on my pc

### Project file & Debug log uploads

[debug_Sun_Nov_03_18_30_11_30596.log.zip](https://github.com/user-attachments/files/17611639/debug_Sun_Nov_03_18_30_11_30596.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

AnthonyZDavis, 2024-11-05:
### **OrcaSlicer Version**
2.2.0

### **Operating System (OS)**
Windows 11 Home

### **OS Version**
23H2

### **Additional system information**
- NVIDIA GeForce GTX
- Intel Core i7 8th Gen

### **OrcaSlicer Usage**
- Kingroon KP3S 3.0
- Anycubic Kobra Neo
- Bambu Lab A1 Mini

### **How to reproduce**
Downloaded the installer from GitHub
Run as admin

### **Outcome**
Execution blocked by Microsoft Defender SmartScreen

### **Outcome Screenshot Upload**
[image]

### **Possible reasons**
Microsoft Defender SmartScreen False Positive
Antivirus False Positive
Missing signature

ScrapyCocos, 2024-11-12:
I found the issue! 

For anybody having the same issues with the installer not lunching at all, click right on the installer after you downloaded it, select properties and tick the box down the bottom where it says Unlock.
https://www.youtube.com/watch?v=K_3MpI-i5Fc

closed 2024-11-12 by ScrapyCocos