OrcaSlicer/OrcaSlicer #7359: Web Interface - Ability to control OrcaSlicer from a web browser, upload files from device to project, etc

MistressPlague, 2024-11-03:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Ish - There's no use of orcaslicer for mobile users or people not near a computer. A web interface would solve that entirely.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It should host the same ui as on computer onto a web interface, based on capture of its own window for example, and allow full control of it through said web ui. It should have a added handler for drag and drop as well as a overlay upload button to upload files to the currently open project. This would open up using the slicer entirely, even on a phone.

### Additional context

I can assist further with technical ideas, as I do programming myself, feel free to ask for elaborations.

MistressPlague, 2024-11-03:
Effectively the software would act as the server, and serve a image as the html, and wherever clicked is then sent to server, where it will figure out what control to click on, and will programmatically click on that control.

Do not use windows api level clicking, as that would inhibit use of said computer while the slicer is in use. This opens up someone else using said computer while someone else in the household could slice something.
While the web ui is in use, it would also be worthwhile to suppress any focus calls or similar so the orca is effectively silent on the computer, not disturbing the computer, but instead using the js alert function for notifications, etc.

MistressPlague, 2024-11-03:
So, to summarize the steps for a initial build of this, this is what is needed to be done:

1. Host a TCP server for the web interface
2. Reply with a window capture of orca slicer
3. Have JS on the client side sent the x & y of where they selected, and send both clickdown and clickup to accord for hold clicking  / dragging
4. Handle such on server, if down and up was within a tolerance of time, click the control at the location, otherwise calculate the start and end location and handle such on server.

Potential expandable parts:
- You could silence right click for the client browser, and send right click and build up a virtual right click menu or something.
- You could implement more camera handling support, by having a "web ui friendly" control scheme for camera controls.
- You could have hold click act as right click if the top of this section isn't feasible.

MuldersMakes, 2024-11-04:
wouldnt this be as viable? https://github.com/linuxserver/docker-orcaslicer

B4sicallyF0x, 2024-11-04:
> wouldnt this be as viable? https://github.com/linuxserver/docker-orcaslicer

That would work, but people would need to install docker and then orcaSlicer. The idea is to have an integrated web server on the actual orcaSlicer, so it's more accessible for people to use it, wasting less resources

MuldersMakes, 2024-11-04:
> > wouldnt this be as viable? https://github.com/linuxserver/docker-orcaslicer
> 
> That would work, but people would need to install docker and then orcaSlicer. The idea is to have an integrated web server on the actual orcaSlicer, so it's more accessible for people to use it, wasting less resources

its bloated and lags as it is, with only really having 4 workable instances already, doesnt need more bloat.

B4sicallyF0x, 2024-11-04:
> its bloated and lags as it is, with only really having 4 workable instances already, doesnt need more bloat.

Idk, works fine for me on desktop

MuldersMakes, 2024-11-04:
> Idk, works fine for me on desktop

with 4-6 instances?

B4sicallyF0x, 2024-11-04:
> with 4-6 instances?

I never needed 6 instances, but I'll try later and will let you know

B4sicallyF0x, 2024-11-14:
> with 4-6 instances?

Just tried. 6 OrcaSlicer instances opened with projects loaded, works perfectly. They do took a long time to boot, but once opened, they work just fine. Using OrcaSlicer 2.1.1 on a Intel i7-1255U with 16GB RAM

closed 2025-02-20 by github-actions[bot]

pank85, 2026-10-21:
Hello, 
I vibe coded a simple ui just to upload and slice. No extra options at the moment.
You can send the sliced gcode straight to you klipper printer.

https://github.com/pank85/print-portal