OrcaSlicer/OrcaSlicer #7367: Anycubic PETG profile has 0g/cm³ weight

avxmw, 2024-11-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Debian 12

### Printer

Anycubic Kobra 3

### How to reproduce

1. Select "Anycubic PETG"
2. Slice
3. See correct amount in meters, but 0 weight

### Actual results

Weight is being shown as 0.

### Expected results

Should show (somewhat) correct weight. My googling suggests 1.27g/cm³ would be a/the value for PETG.

### Project file & Debug log uploads

[Spänesammler2-bugged.zip](https://github.com/user-attachments/files/17616833/Spanesammler2-bugged.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[image]

MxBrnr, 2024-11-04:
You can enter this information yourself into the filament profile and then save it.
[image]

avxmw, 2024-11-04:
I know and did, thank you. But since this profile got shipped with the installation I'd think it should also be fixed properly and thus I filed the bug.

PR #8209 (2025-01-28): QOL fix : Removed filament_density: 0 from Kobra 3 profiles (fixes #7367)
# Description

This removes filament_density: 0 setting from Kobra 3 profiles (fixes #7367)
commit 1fc64c55 (2025-01-28): QOL fix : Removed filament_density: 0 from Kobra 3 profiles (fixes #7367) (#8209)

Removed filament_density: 0 from Kobra 3 profiles (fixes #7367)
closed 2025-01-28 by Noisyfox