OrcaSlicer/OrcaSlicer #7368: top/bottom solid infill overlap BUG

chemlife, 2024-11-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu lab P1S

### How to reproduce

On highest TOP surface overlap is working as it should.
But on any other middle TOP surfaces orca slicer uses default value not the one you enter.
Make a staircase kinda 3d model and slice it with TOP/BOTTOM solid infil overlap set to 80-120%. You can see only the top surface is using corect overlap, other top treated middle surfaces are using some default value.

### Actual results

[image]
[image]
[image]

### Expected results

There should be entered overlap on all top solid infil surfaces not just on the highest TOP surface. 

### Project file & Debug log uploads

[Part1.zip](https://github.com/user-attachments/files/17617572/Part1.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

discip, 2024-11-05:
@chemlife
Try using **`Only one wall on top surfaces`**.

[image]
<br>
Also I suggest to use this instead of **`Monotonic`** for better top surfaces:

[image]

closed 2025-02-12 by github-actions[bot]

henrivdr, 2025-02-12:
@SoftFever  it's a pitty this is not going to be addressed since those are all legitimately top surfaces, right?