OrcaSlicer/OrcaSlicer #7382: printing in air and not identifying the need for support

ycyclop, 2024-11-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

MaxOS 15.1

### Printer

Creality ender 3 KE

### How to reproduce

Slice the attached file. see that in layer 3 it start printing
1:  A very long line that does not have support in layer 1 or 2 (which means it will drop to bed level) between two blocks
2:  A shorter line that does not have an end to attach too which means it will fall to bed level.
Setting Support (tree or normal) will not create a support for these lines

### Actual results

In the second case for example, since the end does not have support not not second anchor the line simply moves freely when the printer head moves away as can be seen in the attached photo
[image]

### Expected results

The slicer should notify that support is needed if user does not select this option, and add support when the option is selected (adding the correct support)

### Project file & Debug log uploads

[NoSupportBug.3mf.zip](https://github.com/user-attachments/files/17631219/NoSupportBug.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

buzzhuzz, 2024-11-05:
I've met this problem myself previously. My understanding is that support generation heuristics works based on bridged surface area which is zero in this case. In addition to that, surface built at higher layers stands above support generation threshold angle.

Workarounds are:
1. manual support paint
2. add chamfer (even 0.5mm should be enough) in your design to automatically trigger support generation

I've stick to second workaround and actually forgot about this bug. Looks like it is time to properly fix this problem.

ycyclop, 2024-11-05:
Thank you for the quick reply
1: Tried the manually support. didn't work as expected. from what I saw the tree support was generated but did not support the edge. it did support the side where I marked the support but it seemed to miss the edge, so it was still floating in air
2: This is not my design so adding a chamfer to the stl will not be that easy (never tried that) with the tools I have

buzzhuzz, 2024-11-06:
It is possible to configure OrcaSlicer to auto generate supports for this model:
[image]

To achieve that you need to enable normal supports since only normal supports has "Normal support expansion" config. That config makes support wider than polygon it supporting by set amount. 

Additionally, some extra tweaks needed for this specific model:
1. Rotate interface pattern by 45 degrees
2. Set top Z-distance to 0.

closed 2025-02-13 by github-actions[bot]