OrcaSlicer/OrcaSlicer #7387: AppImages not running

TheRealOne78, 2024-11-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

V2.2.0, V2.1.0

### Operating System (OS)

Linux

### OS Version

Gentoo x86_64 linux 6.11.5

### Additional system information

```
                                           ​ OS: Gentoo Linux x86_64 
                                           ​ Host: ASUSTeK COMPUTER INC. PRIME B360M-K 
                                           ​ Kernel: 6.11.5-gentoo 
                                           ​ Uptime: 5 hours, 57 mins 
                                           ​ Packages: 1473 (emerge) 
                                           ​ Shell: zsh 5.9 
                                           ​ Resolution: 1920x1080 @ 60.00Hz 
                                           ​ DE: Xsession 
                                           ​ WM: dwm 
                                           ​ Theme: simplewaita [GTK2/3] 
                                           ​ Icons: pixelitos [GTK2/3] 
                                           ​ Terminal: urxvt 
                                           ​ Terminal Font: MesloLGS NF 
                                           ​ CPU: Intel i7-9700F (8) @ 3.000GHz 
                                           ​ GPU: AMD ATI Radeon RX 6600/6600 XT/6600M 
                                           ​ Memory: 2488MiB / 32025MiB (7%) [-==============] 
                                           ​ GPU Driver: amdgpu 
                                           ​ Disk (/): 39G / 98G (42%) [------=========] 
                                           ​ Font: Sans 10 [GTK3] 
```

### Printer

N/A

### How to reproduce

1. Download and change the appimages to executable.
2. Execute them

### Actual results

```
❯ ./OrcaSlicer_Linux_V2.2.0.AppImage
/tmp/.mount_OrcaSlEpPNtb/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

❯ ./OrcaSlicer_Linux_Ubuntu2404_V2.2.0.AppImage
/tmp/.mount_OrcaSlis53T4/bin/orca-slicer: symbol lookup error: /tmp/.mount_OrcaSlis53T4/bin/orca-slicer: undefined symbol: g_once_init_leave_pointer
```

### Expected results

To start the program

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

HakunMatat4, 2024-11-06:
>libwebkit2gtk

This will always be the biggest problem with these slicer.
Since day 1 with BambuLab, this always gave us hell.
Unfortunately, this is a system-wide thing so you cannot just update it without corrupting the entire system.

Have a look on the ticket I just opened, that should help you to get the older version to run again, it is better than no slicer.

TheRealOne78, 2024-11-06:
> Have a look on the ticket I just opened, that should help you to get the older version to run again, it is better than no slicer.

Thanks, but which one?

HakunMatat4, 2024-11-06:
I solved the problem following the ticket 7210.
You have to update that package and set the `PRINTER_IP:PORT`

As for the package I mentioned earlier, there is a package with similar name which you cannot just update, I think I got those mixed.
libwebkitgtk is a content engine library.

sudo nala install libwebkitgtk-6.0-4 libjavascriptcoregtk-6.0-1 ( the first package pulled the second )
Set [IP]:7125 or mDNS works trident01.home.arpa:7125

TheRealOne78, 2024-11-10:
Alright, that kind of works. Tho I rather made some temp softlinks from 4.1 to 4.0. I don't get you got Orca to run with 6.0, since I also have 6.0 installed on my system.

I will close this issue since there's another duplicate issue.

closed 2024-11-10 by TheRealOne78