OrcaSlicer/OrcaSlicer #7388: Filament printing speeds

KarleHeinz, 2024-11-05:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have to change every time the print speeds when I want to print a different material.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add to the "overwrite print settings" the print speeds. 

### Describe alternatives you've considered

Add the values for the printing speed to the filament settings.

MxBrnr, 2024-11-05:
It was done this way on purpose so that you can use the same process and print speeds across multiple filament profiles, and use the same filament profile across multiple processes with different speeds. If you want speed to be specific to each filament, then it would be better to create and save each process, identifying the filament to use it with in the process name.

jbowers01, 2024-11-07:
My solution is to test and then cap the maximum volumetric speed for each filament. For example, the default profiles print TPU at 7.5 mm/s, which I have found to be reliable for my printer based on the filament settings.

tlhintoq, 2024-11-07:
Material really isn't an indication of the speed needed for a model.
One guy prints a 28mm mini with a ton of detail out of PETG - that's one speed.
Next guy with the same material is making big rectangular cases for tools.  Same material but a vastly different speed.

Then you have nozzle size.  a .2 nozzle is not moving at the same speed on a tiny thing, as a 1.0 nozzle on a big thing.
Material shouldn't dictate speed.

jbowers01, 2024-11-07:
That's gonna be your printer profiles then. The material settings just stop me from printing faster than the material is capable of for quality or layer adhesion. I tune every type and brand individually, and then tune specific profiles for the task. The reality is that you're never in danger of printing too slow, so you're tuning the filament to avoid pushing it past its limits and the profile for the part.

closed 2025-02-14 by github-actions[bot]