OrcaSlicer/OrcaSlicer #7393: OrcaSlicer does not respect travel acceleration outside printing movements

HakunMatat4, 2024-11-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0, 2.2.0

### Operating System (OS)

Linux

### OS Version

Mint Cinnamon 21.3

### Printer

Voron Trident

### How to reproduce

1. Print as usual
2. Travel acceleration isn't respect after the purge line.

### Actual results

This problem has been driving me nuts but I couldn't figure out the root cause until today while trying to get the latest OrcaSlicer release to work.
I had to use PrusaSlicer and like magic, after the purge line the toolhead traveled at 6000mm/s instead o 10mm/s like OrcaSlicer does. It is painfully slow.

This has always being a problem when printing many objects, by the time the toolhead reaches the starting point, there is enough oozing to destroy the first layer.
To minimize that, I start from as close as to the purge line as possible. That option isn't always available and affects printing quality.

Notice how PrusaSlicer sets the accel globally at the very start and yet OrcaSlicer has not set it.

[image]

The only time OrcaSlicer uses the acceleration is during the printing and that is wrong.
Basically, OrcaSlicer won't use the values set for non-printing movements like after purge line or PAUSE/START while PrusaSlicer set those globally so any movement will use the values set.

[image]

### Expected results

Respect the values of acceleration set for ANY movement, not just during printing.

### Project file & Debug log uploads

[logs.zip](https://github.com/user-attachments/files/17646420/logs.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

igiannakas, 2024-11-06:
That is odd...

[image]

For me Orca sets the acceleration immediately before the first XY move after the print start macros. Could you post a project file?

HakunMatat4, 2024-11-07:
I checked everything again, both slicers do have the same virtual travel, but yet, both behave differently.
That travel tho is not registered.

Whatever is happening, lies here, there is no accel set here, OS and PS are using different values in my case.

```
G1 Z.65 F30000
G1 X164.475 Y158.947
```

OrscaSlicer
[image]

PrusaSlicer
[image]

Ergonomicmike, 2024-11-07:
I'm not a coder, nor dev, etc. So I don't know what "Best Practice" is for Orca/Klipper. But after floundering around in my first ever dive into Klipper last year, I found it best to move the Line Purge part of START_PRINT to a separate call in Filament (Materials)>Advanced>Filament start G-code. 

```
 ; filament start gcode
_LINE_PURGE_TPU
```

This solved a lot of problems that I was having by burying the Line Purge in START_PRINT. 

And, as a bonus, this allowed me to call different Line Purge macros for different materials. (TPU needing a much slower extrusion speed than PLA, for example.)

I'm not sure if I should call my method a "work around" if this isn't a bug in Orca. Regardless, it might solve the OP's instant issue.

HakunMatat4, 2024-11-07:
> And, as a bonus, this allowed me to call different Line Purge macros for different materials. (TPU needing a much slower extrusion speed than PLA, for example.)

That alone explains why your solution isn't a workaround but an actually nice solution.
Why does your macro starts with underscore?? 

In my case tho, the entire leveling and all happens with the nozzle at 150c, only then it parks at the front left corner to fully heat the nozzle.
I do that to avoid oozing and it works wonderfully.
I wonder if the printer will have those values in place in time, based on your experience, that shouldn't be an issue.

```
    ## Purge line
    G92 E0
    G1 X3 Y20 Z0.3 F5000.0
    M109 S{hotendtemp}      ;start heating hotend and WAIT
    G1 X3 Y120.0 Z0.3 F2000.0 E15
    ## End purge line
```

I will try to create a `[PURGE_LINE]` macro and set it like you did, hopefully, that will solve the problem.
The problem tho is that OrcaSlicer prior to the actual print, it does not use the accel values.
It is worth a try.

HakunMatat4, 2024-11-07:
The above is actually an old code, this was my current code and an attempt to fix it: The last line.
PS sees that while OS does not.
I will try the have a purge line macro instead.

```
    G92 E0
    G1 X3 Y20 Z0.3 F5000.0
    M109 S{hotendtemp}      ;start heating hotend and WAIT
    G1 X3 Y125.0 Z0.3 F2000.0 E15
    G0 X3 Y130.0 Z0.3 F6000
```

Ergonomicmike, 2024-11-07:
The underscore before the macro name is, [per Klipperscreen documentation](https://klipperscreen.readthedocs.io/en/latest/macros/), to hide macro names from the Interface. (I don't see a need to call Line Purge from either Mainsail or Klipperscreen.)

Hope my "workaround" works for you.

HakunMatat4, 2024-11-07:
@Ergonomicmike , your approach although it is nice to know, it did not solved my problem :(
Ohh yes, I forgot that detail to hide it from KlipperScreen, thank you.

I recorded both prints from both slicers.

OrcaSlicer does process `G0 X3 Y130.0 Z0.3 F6000` but from there to the center of the bed, it does not use my travel settings.
https://github.com/user-attachments/assets/5b6c3b26-5c15-41ec-81cf-5cc788827619

PrusaSlicer on the other side is day and night difference.
https://github.com/user-attachments/assets/341d57e0-7839-4e07-9e46-95eda116c8f9

Ergonomicmike, 2024-11-07:
Sorry it didn't fix your instant issue. 

AFAIK, I don't have this issue. Since you have the issue, and since it seems to be following the Slicer, my WAG is that your ultrahigh settings (compared to mine (Ender 3 v2)) hits/triggers some kind of limit in Orca. (Kinda the way that volumetric speed limit can slow print speed down.)

But the dev's here are very good at solving puzzles. I'll watch as they figure this one out.

(As a small aside (and it probably doesn't make a difference) but I have 
 `G92 E0 ; Reset Extruder`
at the end of my Line Purge macro. You don't.)

HakunMatat4, 2024-11-07:
> You don't

It is within `PRINT_START` macro after the purge line.
Those 2 prints above were performed as it is ( after I did the PURGE_LINE test ) so that is not it.
So both slicers are using the same `PRINT_START`, the same macros, same speeds but different behavior.

For many people this isn't an issue, when printing lots of objects, ASA-CF at 260c and so on, this creates annoying headache with the oozing getting stuck on the first layer.

For this alone, I would rather use PS.
You have no idea for how long I have been trying to solve this, that is why the line `G0 X3 Y130.0 Z0.3 F6000` with the intention of making OS to use the latest travel speed loaded but nope.

I use OS since the very first release when we ( Linux folks ) had to manually compile our slicer.
I always encountered occasions where OS would not behave as expected driving me nuts.
One time it was not triggering the retraction so my prints were looking like dogsh1t, countless hours and filament wasted thinking that it was my printer, long story short it was indeed OS misbehaving and required a hotfix.

That was one example, there were others so I know that this is not me but OS doing.

```
    ## Purge line
    G92 E0
    G1 X3 Y20 Z0.3 F5000.0
    M109 S{hotendtemp}      ;start heating hotend and WAIT
    G1 X3 Y125.0 Z0.3 F2000.0 E15
    G0 X3 Y130.0 Z0.3 F6000

    G90 ; absolute position mode
    G92 E0 ; Reset Extruder
    M83 ; use relative distances for extrusion

    STATUS_BUSY # LED Effect
```

Ergonomicmike, 2024-11-07:
Another WAG: I've noticed that Linux users sometimes have problems that Windows users don't with Orca. Can you run the portable Window version of Orca in a VM to slice your model just to see if it's a Windows/Linux issue?

igiannakas, 2024-11-07:
> I'm not a coder, nor dev, etc. So I don't know what "Best Practice" is for Orca/Klipper. But after floundering around in my first ever dive into Klipper last year, I found it best to move the Line [...]

This is fine as long as you don’t use multi material printing - if you do, this is called every time the filament swaps.

igiannakas, 2024-11-07:
Any chance you can post the two gcode files? @HakunMatat4 

I haven’t personally noticed this but, I do reset the printers acceleration to the default value in printer config at my print end macro, so I think that one is used.

a workaround would be to set manually an acceleration value in your print start macro right after the purge line. This would set the correct accel for the first travel move and then the slicer would take over and replace it.

but do please post the gcode files from orca and prusa so I can take a closer look if you can

HakunMatat4, 2024-11-07:
> a workaround would be to set manually an acceleration value in your print start macro right after the purge line. This would set the correct accel for the first travel move and then the slicer would take over and replace it.

@igiannakas This is what I have in place but OrcaSlicer refuses to stick to it lol

```
    ## Purge line
    G92 E0
    G1 X3 Y20 Z0.3 F5000.0
    M109 S{hotendtemp}      ;start heating hotend and WAIT
    G1 X3 Y125.0 Z0.3 F2000.0 E15
    G0 X3 Y130.0 Z0.3 F6000
```

I sent the log and both gcodes when I opened the ticket lol
The only time OS will use my speeds for the first time is at the line 1468.
PS calls it at the line 782 but it respects the `F6000` above.
I don't know what is going on haha

Thanks man

[logs.zip](https://github.com/user-attachments/files/17659100/logs.zip)

igiannakas, 2024-11-07:
Do me a favor, sorry I haven’t checked the zip file yet but does it also include the orca project? im suspecting I don’t see this because I may be using a skirt.

Ergonomicmike, 2024-11-07:
I always use a skirt too. Perhaps @HakunMatat4 can test to see if a skirt changes what he's reporting without one?

HakunMatat4, 2024-11-07:
> test to see if a skirt changes

Nope, it still moves slow and the best proof that my eyes aren't failing on me, the motor sound.

The gifs I shared above, there are their respective videos. 

Noticed how OrcaSlicer is a more thicker and louder sound due to slow motor movement ( that is also why I have my speeds as high as possible because slow movements means too much resonance. That is also why I don't use that new seam because you are forced to speeds around 50mm/s, big pass for me. )

Noticed how PrusaSlicer is a much finer and quieter motor sound, the faster I go, the motor "silent" the motor gets. 
If that matters, this is a LDO Trident 300mm Cube.

The skirt test looked as slow and as loud as before.
I know this is a first world problem but....

-----

And again, I only noticed PS behavior as I expected OS because I have not touched my printer in ages so I was checking the first layer to make sure Cartographer probe ( virtual z_home ) was still holding the offset. 
Normally, I just hist Upload and Print, I don't babysit my printer so why I absolutely love this printer.

This is the gcode and project file with the skirt in place.
[projects.zip](https://github.com/user-attachments/files/17670127/projects.zip)

HakunMatat4, 2024-11-12:
> If you set first layer accel to 0 orca will ignore all acceleration values except outer and inner wall speed and _all moves will be done with the last one set_, even infill and travel moves will be done with outer/inner wall acceleration(depending on what the slicer did last).

You might have found the problem....

I have read your comment and my first layer acceleration is set to 500mm/s while on PS is done at 1000km/s
But PS moves so much faster to be just 500mm/s missing.
I will do some test and post here.

[image]

>If you set first layer accel to 0 orca will ignore all acceleration values except outer and inner wall speed and all moves will be done with the last one set, even infill and travel moves will be done with outer/inner wall acceleration(depending on what the slicer did last).

Look, I like OrcaSlicer but I have learned the hard way to never trust its "auto" stuff.
Always, always, always set values instead of letting the slicer logic to do it for you.
This is why even so OrcaSlicer allegedly supports z offset, I have one printer profile for each PEI sheet and if PLA or ASA.

I don't wanna the slicer sending the nozzle into the sheet because its logic went RIP.
This has happened with PA, retraction logic, wasting so much filament and days only to find out it was a logic problem so I have values set everywhere I can, and I let klipper to process as much info as possible,
Fool me twice, no more lol

[image]

Ergonomicmike, 2024-11-12:
> off topic but what I do with different plates and materials is add code like the below to the **print start** gcode section. **Not the filament gcode (as this is run every time there is a tool change with ERCF).**
> 
> ```
> {if (curr_bed_type=="Textured PEI Plate" || curr_bed_type=="Textured Cool Plate")} 
> ; Gcode offset for Textured plate for all materials
> SET_GCODE_OFFSET Z_ADJUST=-0.045 MOVE=1
> {endif}
> 
> {if (filament_type[current_extruder]=="ABS" || filament_type[current_extruder]=="ASA")}
> ; gcode offset for high temp materials to reduce squish on all plates
> SET_GCODE_OFFSET Z_ADJUST=+0.025 MOVE=1 
> {endif}
> ```

Okay, I now realize that adding gcode in "filament" is bad form (if I ever use multiple filaments). And I'm glad to see that conditionals are supported in the "print start" field. So, even more off topic, not being a coder, could I hack an "if" statement to look to see if I'm using TPU so that I can use a custom LINE_PURGE macro? (I don't see how Orca itself knows what type of filament I'm using. Can it read my description?)

Ergonomicmike, 2024-11-12:
> There was a PR to make it use the travel acceleration for the first layer, limiting the first layer acceleration for print moves only (#6253). Also in that PR it was proposed to set another option to set travel acceleration for the first layer as a % of the global travel acceleration, but they were rejected from what I understand as quoting Softfever, "The reason for using first layer jerk and acceleration for travel is that rapid travel increases the chance of the nozzle picking up freshly laid filament on the first layer based on user feedback."

I saw that report from Softfever. I don't understand it. Everyone knows that toilet paper tears cleanly when acceleration and speed are high. It seems to me that slow travel speeds would give time for ooze to bond to the first layer sections and pull.

igiannakas, 2024-11-12:
Yes you can set the conditional like this based on the material type selected in the filament - material type drop down.

if - tpu - run the tpu purge
Else run the normal purge 

the statements are like this:

{if filament_type[current_extruder]=="TPU"}
TPU_line_purge_macro
{else}
line_purge_macro
{endif}

Ergonomicmike, 2024-11-12:
Tnx. I don't want to hijack this (now soon to be closed?) thread.

~~I wasn't aware that Orca could grep (?) for filament names in my descriptions. Especially since there is no System Present for TPU for my Creality Ender 3 v2. So, if I'm understanding correctly, I could name a filament as TPU95, another as TPU85 and 
`{if filament_type[current_extruder]=="TPU95"}`could act on it?~~

Oops. I just now saw, for the first time, that there is a hard setting for "Filament Type" in Materials in Filament. D'uh. I understand now. 

Thanks again.

igiannakas, 2024-11-13:
Leave the ticket open as if this fixes it for you it may be a test case showing the need  for the feature.

in general from what I understand, the principle is to keep the slicer as simple and straight forward as possible, to avoid hundreds of options like super slicer that complicate matters when merging upstream code etc but offer features that have demonstrated use cases/benefits

the issue with FOSS is that everyone here does it as a side of desk thing and there are hundreds of things to do but only so little time to do them :) sometimes I wish this could be my day job as I personally love coding for this but it won’t pay the bills so… :(

HakunMatat4, 2024-11-13:
>  I wish this could be my day job as I personally love coding for this but it won’t pay the bills so… :(

Sure thing, I'll leave it open.

I'm in IT but I hate coding, yet, I had to deal with python and I absolutely adore python now haha
I found it simple and somewhat fun to work with.

closed 2025-02-20 by github-actions[bot]