OrcaSlicer/OrcaSlicer #7397: Segmentation fault (core dumped)

vldmr-d, 2024-11-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

OrcaSlicer_Linux_Ubuntu2404_V2.2.0.AppImage

### Operating System (OS)

Linux

### OS Version

Linux Mint 22 (Wilma)

### Additional system information

Intel(R) Core(TM) i7-6800K CPU @ 3.40GHz
32 RAM
NVIDIA GeForce GTX 1070 Driver Version: 550.120

### Printer

Sapphire Plus - Klipper

### How to reproduce

```
~/Desktop$ ./OrcaSlicer_Linux_Ubuntu2404_V2.2.0.AppImage

[2024-11-06 18:09:54.268666] [0x000070f9650e9480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:142496): Gtk-CRITICAL **: 18:09:55.015: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:142496): Gtk-CRITICAL **: 18:09:55.015: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:142496): Gtk-CRITICAL **: 18:09:55.015: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
....................................
(orca-slicer:142496): Gtk-CRITICAL **: 18:09:56.594: gtk_widget_set_size_request: assertion 'width >= -1' failed
No such file or directory

(orca-slicer:142496): Gtk-CRITICAL **: 18:09:57.616: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:142496): Gtk-CRITICAL **: 18:09:57.616: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:142496): Gtk-CRITICAL **: 18:09:57.623: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:142496): Gtk-CRITICAL **: 18:09:57.623: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:142496): Gtk-CRITICAL **: 18:09:57.626: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:142496): Gtk-CRITICAL **: 18:09:57.629: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:142496): Gtk-CRITICAL **: 18:09:57.630: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Segmentation fault (core dumped)
```

### Actual results

Starting, waiting, Segmentation fault (core dumped)

### Expected results

Work correctly

### Project file & Debug log uploads

```
[error]	2024-11-06 17:35:50.385826[Thread 0x0000774ab0a26480]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-11-06 17:35:50.397633[Thread 0x0000774ab0a26480]:Unable to open the file/home/user/FreeCad/STEP_FILES/Dremel_stand-Body015.step
[error]	2024-11-06 17:35:50.397651[Thread 0x0000774a44a006c0]:Unable to open the file/home/user/FreeCad/STEP_FILES/Dremel_stand-Body016.step
[warning]	2024-11-06 17:35:50.816967[Thread 0x0000774ab0a26480]:17:35:50: Warning: Error running JavaScript: file:///tmp/.mount_OrcaSlqH9ZVD/resources/web/homepage/index.html?lang=en_US:1:9: ReferenceError: Can't find variable: __wxOut0
[warning]	2024-11-06 17:35:50.819766[Thread 0x0000774ab0a26480]:17:35:50: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-11-06 17:35:50.984960[Thread 0x0000774ab0a26480]:17:35:50: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-11-06 17:35:50.986513[Thread 0x0000774ab0a26480]:17:35:50: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-11-06 17:35:52.040971[Thread 0x0000774ab0a26480]:get_version, get_version not supported,return 00.00.00.00!
[error]	2024-11-06 17:35:53.394434[Thread 0x0000774ab0a26480]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-11-06 17:35:53.559311[Thread 0x0000774ab0a26480]:calc_exclude_triangles:Unable to create exclude triangles
```

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2024-11-07:
See #7210

vldmr-d, 2024-11-07:
Changed "print_host": "192.168.10.93" to
"print_host": "192.168.10.93:7125"
and it seems to work

[image]
[image]
[image]

formatCvt, 2024-11-17:
thanks, adding port fix issue in my case

lkarlslund, 2024-11-17:
Same issue here running both 2.2.0 and latest action build Ubuntu 24.04 AppImage on Ubuntu 24.10

The fix by adding port 7125 in the json files is a functional workaround for now

closed 2025-02-24 by github-actions[bot]