OrcaSlicer/OrcaSlicer #7402: 🚀 Feature Request: Option to Maintain Uninterrupted Top Surface Pattern Despite Overlapping Objects

JerryblackVe, 2024-11-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Description:

I am using Orca Slicer and would like to request an option that allows the top surface to be printed as a complete, uninterrupted pattern, even when there are other objects—such as text, numbers, or figures—overlapping the top surface.

Current Behavior:

When the top surface has overlapping elements (like letters or numbers), Orca Slicer tends to create significant anchoring or pronounced adjustments on the top surface where these elements are present. This results in the surface pattern being interrupted, leading to a less smooth and aesthetically pleasing finish.

Example:

Model: A flat top surface with raised letters or figures on it.
Issue: The slicer modifies the top surface pattern beneath the overlapping elements, causing interruptions and visible artifacts in the print.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Option Addition: Introduce an option that allows the top surface to be printed continuously without interruptions, regardless of any overlapping objects above it.
Benefit: This would prevent the slicer from altering the top surface pattern due to overlapping features, resulting in a smoother and more complete finish.

### Describe alternatives you've considered

Benefits:
Improved Aesthetics: Enhances print quality for models requiring a consistent top surface texture.
User Control: Provides users with greater control over the surface finish of their prints.

### Additional context

This feature would be especially beneficial for prints that include decorative text or intricate designs on the top layer.
It would help maintain the integrity of the surface pattern, ensuring that overlapping objects do not interfere with the overall appearance.

tlhintoq, 2024-11-07:
More realistically... and the way we've done it for years... and probably faster than the developers adding it...

Split the object at the point where you want the slicer to see it as a continuous surface. Let it see the bottom part of the *assembly* as one part with a flat top and the top part as a separate part.  Just make the model into what you want the slicer to treat it like.

The problem with the software making that decision arbitrarily is that not every person will want that, nor should every object just be seen that way.  Sometimes I have a hollow thing with lettering (or whatever) and it needs to be seen as one object, no layers in the middle, so it can be filled later, or so a screw can go through without hitting a layer.  Software can't _just know_ where its right or wrong to do that.

JerryblackVe, 2024-11-07:
This feature exists in other slicers; in Cura and Creality Print, it's called "Top Surface Skin Layer."

I was wondering if it would be possible to implement something similar in Orca Slicer, as I have needed this function many times and believe it could be useful to many others as well.

THZK100SC, 2024-11-07:
This feature is very good. I also have a problem with it sometimes. Personally, I sometimes find it helpful to use a classic wall generator. 

Personally, I am in favor of adding this feature. It would make it easier. :-)

HakunMatat4, 2024-11-07:
Do you guys have a pic or something showing this??

THZK100SC, 2024-11-07:
I can't seem to accurately simulate the problem. But the point is that the font sometimes has anchors to the last top layer. And then the upper layer is not whole, but has a torn structure. The video shows the result, which is great. 

https://github.com/user-attachments/assets/1eec4337-bfe5-4d20-94a9-25951e21bcb2

HakunMatat4, 2024-11-07:
> I can't seem to accurately simulate the problem. But the point is that the font sometimes has anchors to the last top layer. And then the upper layer is not whole, but has a torn structure. The [...]

Ahh yeah, it will stop the topmost layer to print the letters.
This would be a very nice to have indeed and that would also reduce printing time without having to slowdown.

The only catch is that it should only be enabled with letter, anything functional printed on top of it might not have a solid adhesion because it will be like bridge with support surface.

But very nice to have 10/10

Noisyfox, 2024-11-07:
I had this idea in mind for a long time, however I haven't find a good way of implementing this, mainly because it's hard to set a good condition for when and where this should be applied on the model. For example you definitly don't want to apply this to a shallow slop just because it has top surface. Might be able to add this as a modifier. Or a threshold of the ratio between the surface area of top & non-top area. Or she surface area of the place that get covered by top layers etc.

JerryblackVe, 2024-11-07:
Here are reference images to illustrate the issue:

In the first image (Layer 1), we see a continuous pattern on the top surface. However, in the second image (Layer 2), parts of the letters are already visible, even though they should only start appearing in the third layer (as shown in the third image). Orca Slicer begins showing these details from Layer 2, and currently, there is no way to prevent this from happening.

It would be helpful to have an option that allows users to maintain a fully continuous top layer pattern until the designated layer for overlapping elements, avoiding interruptions in the pattern due to premature detailing.
[image]
[image]
[image]

kylek29, 2024-11-07:
I completely agree with this. It's one of those things that annoys me (if I'm going for a good finish) and I think having it has an option would be great.

JerryblackVe, 2024-11-07:
The purpose of this improvement isn't just for letters; it's for any object that overlaps the top surface, ensuring that the pattern remains undisturbed

JerryblackVe, 2024-11-07:
Here’s another example:

In image 1, there’s a hexagon on the top surface of a square. I’d like the top surface pattern of the square to remain uninterrupted by the hexagon’s fill. Ideally, there should be an option to print the hexagon directly on the top surface without disrupting the square’s pattern. This would result in a more uniform and cleaner surface finish.
[image]
[image]
[image]

jeremytodd1, 2024-11-24:
This issue has plagued me since Orca released.
There is currently no way to tune for this and it makes top layers for signs and maps look a lot worse than they could.
I'm all for getting a fix for this implemented as it has caused me quite a bit of issue the past 2 years.

Yishher, 2024-12-13:
tThis also affects infill modifiers, making them substantially weaker, top surfaces/internal solid infill in general have a lot of weird bugs in my experience, makes the slicer unusable for some parts. Would love for this to get fixed, only real problem with the slicer I've experienced. For anyone reading this try experimenting with "ensure vertical shell thickness", "minimum sparse infill threshold", "only one wall on top surfaces", and "one wall threshold" sometimes these settings can help, but nothing seems consistent and it changes with different files.

https://github.com/user-attachments/assets/3929fa6f-1e95-41d8-a9c0-36501ab9f400

CodyPrince, 2025-03-20:
Searching for my a fix I found this thread. I would really love to see this feature in the future. Hope it isn't a dead option.

JerryblackVe, 2025-03-20:
Hello, everyone!

Let's push together to see if we can get this feature added in this request. Hopefully, we can have it in future updates of OrcaSlicer. It looks like many people are facing the same issue as I am.

JPFASTLANE, 2025-06-18:
> Hello, everyone!
> 
> Let's push together to see if we can get this feature added in this request. Hopefully, we can have it in future updates of OrcaSlicer. It looks like many people are facing the same issue as I am.

I vote YES

Olivecns, 2025-07-16:
I vote YES

Thanks 👍🏼

CodyPrince, 2025-07-16:
I vote yes

JerryblackVe, 2025-07-16:
I VOTE YES

PatBarx, 2025-07-16:
I Vote YES!!

Andreea-Balan, 2025-07-16:
I vote yes

jeremytodd1, 2025-07-16:
I also vote yes.
I've been running into this issue on Orcaslicer for close to 2 years now, and it effects the majority of my prints. I would love to have a solution for this. It's very apparent if you print things like signs or maps.

inventor7777, 2025-10-17:
I vote yes for this as well. I'd prefer a better surface quality on my RC spoiler which has text on it