OrcaSlicer/OrcaSlicer #7406: Weird gcode generated - totally outside of the print area

szafranski, 2024-11-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Qidi Q1 Pro

### How to reproduce

Trying to slice attached STL file. 

### Actual results

The gcode that is being generated contains an unnecessary path (or paths, depends on the orientation) and extrusions. That prevents the job to be finished on the printer. "Repairing", fixing the STL or remeshing etc., does not help.

### Expected results

Clean gcode should be generated.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17663546/log.zip)

[v6_Cartridge_90deg_Sliding_Door_Mod.zip](https://github.com/user-attachments/files/17663555/v6_Cartridge_90deg_Sliding_Door_Mod.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[v6_Cartridge_90deg_Sliding_Door_Mod.zip](https://github.com/user-attachments/files/17663620/v6_Cartridge_90deg_Sliding_Door_Mod.zip)
[image]
[image]

Ergonomicmike, 2024-11-07:
https://github.com/SoftFever/OrcaSlicer/pull/7330  ?

szafranski, 2024-11-07:
Possibly related/the same. Yes

closed 2025-02-14 by github-actions[bot]