OrcaSlicer/OrcaSlicer #7408: OrcaSlicer 2.2.0 encounters unknown error and terminates seconds after startup

FuzyWazaBear, 2024-11-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

9th I7-9700K
32GB DDR4
2070 RTX

### Printer

Bambu A1/A1M/P1S

### How to reproduce

1. Update to 2.2.0 from 2.1.1
2. Bambu network plugin installs
3. About 5 seconds passes, program is responsive during the 5 seconds.
4. "Unkown Unhandled Exception" Occurs
5. Program is unresponsive and force closes

### Actual results

[image]

### Expected results

Program is expected to return "unkown Unhandled Exception" and force close.

### Project file & Debug log uploads

No Project

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

This is a continuation of #4964 who also had the same issue with no results.

ReinhardC, 2024-11-23:
Same, I updated to 24H2 recently, maybe it's related to that.

SergInBloom, 2025-02-28:
same issue 

[image]

SergInBloom, 2025-03-04:
I've found that this happens right after I log in to my Bambu account. If not, it works.
Any idea how to fix it?

Windows 11 Pro
24H2
26100.3194
Windows Feature Experience Pack 1000.26100.48.0