OrcaSlicer/OrcaSlicer #7410: Pressure Advance Pattern Calibration Test not showing up after slicing

F0R3IGN3R, 2024-11-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro 23H2 3/‎4/‎2024

### Printer

Ender 3 Pro & Anycubic Kobra Neo

### How to reproduce

Run the Pattern Calibration as instructed and after slicing the plate the model only shows the original placement square and the total print time in the gcode is 1m 9s and uses less than 1g..
[image]
[image]

### Actual results

Not showing after  slicing on both printers.
[image]
[image]

### Expected results

[image]

### Project file & Debug log uploads

[Pressure Advance Pattern Test (Ender 3-Pro)(After Slicing).zip](https://github.com/user-attachments/files/17670793/Pressure.Advance.Pattern.Test.Ender.3-Pro.After.Slicing.zip)
[Pressure Advance Pattern Test (Kobra Neo)(After Slicing).zip](https://github.com/user-attachments/files/17670794/Pressure.Advance.Pattern.Test.Kobra.Neo.After.Slicing.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

F0R3IGN3R, 2024-11-08:
Can someone please help me with this issue im just returning back to 3d printing and im not sure if i installed and setup everything right. I know the prior version to 2.2.0 that i had i was able to run the test and slice the project with no problem but after the update its not working for me anymore. Please advice and thank you in advance.

justr0st, 2024-11-08:
@F0R3IGN3R the issue exists and needs a fix. You are not doing anything wrong. Maybe the reason for this bug is a change to PA pattern in 2.2.0 release.

JHAP91, 2025-01-04:
Hi @justr0st and @F0R3IGN3R, do you know if there is any solution to this bug yet? I am having the exact same thing happening to me too

some1one, 2025-01-06:
I might have figured it out. This seemed to be what fixed it for me. Can someone verify that having it on as "Print by Layer" instead of "Print by Object" corrects the issue?

JHAP91, 2025-01-10:
> I might have figured it out. This seemed to be what fixed it for me. Can someone verify that having it on as "Print by Layer" instead of "Print by Object" corrects the issue?

Worked for me too, thanks!

wernight, 2025-03-22:
Print by layer fixed it. Also had to avoid "Inner / Outer / Inner" wall printing order because that crashes Orca.

closed 2025-06-29 by github-actions[bot]