OrcaSlicer/OrcaSlicer #7417: Adaptive Cubic works unpredictably with assembled objects

sneakyjoeru, 2024-11-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-rc

### Operating System (OS)

macOS, Windows

### OS Version

win10, macos 15.1

### Printer

x1c

### How to reproduce

Set several small objects close together, assemble them

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[YellowFinal_181548.3mf.zip](https://github.com/user-attachments/files/17678976/YellowFinal_181548.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

I need this because if I print separate files close to each other their supports conflict.

sneakyjoeru, 2025-02-07:
Still relevant bug

sneakyjoeru, 2025-05-09:
:-(

sneakyjoeru, 2025-08-08:
:-(

sneakyjoeru, 2025-11-16:
no?

Erosion139, 2026-01-12:
I have discovered this bug as well. Hope it gets considered again!

pgifford, 2026-01-14:
It seems ok in the nightly build I installed a day or two ago.  I didn't change any settings, just opened it and sliced.

Screenshot taken at layer 38, Z = 3.80 mm.

[image]