OrcaSlicer/OrcaSlicer #7419: Support cotact layer - new type

locki-cz, 2024-11-08:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Perimeters above support are printed in air ... sometimes it doesnt hold and print is ruined.  
[image]
[image]

In Superslicer is contalc layers generated on angle, which works on both directions:
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add new type of contact layer support, or possibility change angle - then we can set it to 45 degrees.

buzzhuzz, 2024-11-10:
Give a try to "snug" support type.
There are also "pattern angle" item in advanced section of supports settings: it also may help you in this  case.

locki-cz, 2024-11-15:
> Give a try to "snug" support type. There are also "pattern angle" item in advanced section of supports settings: it also may help you in this case.

new update:

[image]

this is automatic snug support is ok but ... outside perimeters are in air ... view from bottom:

[image]

closed 2025-02-21 by github-actions[bot]