OrcaSlicer/OrcaSlicer #7424: Skirt not maintaining loop count across layers in recent updates.

JagustiC, 2024-11-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

Sonoma 14.3.1

### Printer

All printers (RatRig V-Minion, Crealitys K1, Ender3V2, Ender3V3)

### How to reproduce

1. Set skirt loop count to 3 in the skirt settings.
2. Slice the model and confirm in the preview that the first layer has 3 loops but subsequent layers show only 1.
3. Run the print, confirming only the first layer has 3 loops.

### Actual results

Only the first layer of the skirt has 3 loops; all other layers print a single loop

### Expected results

The skirt should print with the specified loop count on all layers.

### Project file & Debug log uploads

[3DBenchy_SKIRTLOOPS.3mf.zip](https://github.com/user-attachments/files/17688618/3DBenchy_SKIRTLOOPS.3mf.zip)
[log.pdf](https://github.com/user-attachments/files/17688627/log.pdf)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]
[image]

locki-cz, 2024-12-28:
Oh ... still not fixed ... I returned to 2.1.1 because 2.2.0 is full of anoing bugs :(

m00dawg, 2025-01-31:
I don't have much to add other than to say I experienced this as well and will be rolling back to 2.1.1 as well. Without the multiple layers for all loops, I get crud on the build plate since not all the plastic releases from the thin skirt and it's a huge pain to clean (it's the reason I have 3 layers worth of loops).

tetsu97, 2025-02-13:
Same problem. Any idea when it will be fixed? I am building an IDEX printer and would like to start using it with the new version.

Noisyfox, 2025-02-21:
https://github.com/SoftFever/OrcaSlicer/pull/5786

locki-cz, 2025-02-21:
> [#5786](https://github.com/SoftFever/OrcaSlicer/pull/5786)

this function destroyed our standard skirt 3 loops and 3 layer height for easy remove??? :( It is sometimes very hard remove from some sheets just on layer, this is why we used 3x3 for skirts .... for tune first layer and easy remove, can it be somehow revertec back and implemented skirt some loop count on high object better way?

JagustiC, 2025-02-21:
> this function destroyed our standard skirt 3 loops and 3 layer height for easy remove??? :( It is sometimes very hard remove from some sheets just on layer, this is why we used 3x3 for skirts .... [...]

Same here, this is so messed up that not just skirts are hard to remove, but draft shields on high temp materials and big tall prints just can’t hold their own weight and get deformed by heat.

Please bring back the old settings where skirt loop count will stay through the hole skirt/draftshield and not only for first layer. 
Or add an option to chose first layer loop count and above first layer loop count. 

Thanks.

thunderkeys, 2025-02-23:
Any chance of this being corrected with the 2.3.0 release?

JagustiC, 2025-03-04:
> PR raised above.

Thank you for your help here, and all other contributions to OrcaSlicer you’ve made in the past. 

🙏

PR #8562 (2025-03-08): Introduce option to control one wall draft shield
# Description

Introduce option to control whether one wall or all skirt walls are extruded from the second layer onwards.

Fixes #7424

# Screenshots/Recordings/Graphs
(edit - screenshots below updated following latest code commit after Softfever's feedback).

[image]

[image]

[image]

[image]

## Tests
Tested with option enabled & disabled
Tested with multi color print
Tested with draft shield on/off
closed 2025-03-08 by SoftFever