OrcaSlicer/OrcaSlicer #7428: Slowed overhang disables arc fitting for the entire layer

malventano, 2024-11-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

10

### Printer

X1 Carbon

### How to reproduce

Slice any model with "Arc Fitting" and "Slow down for overhangs" enabled. Sample attached using all defaults except infill disabled to make step count jump more clear.

### Actual results

Note that any layers containing even a slight slowed overhang have significantly more gcode steps for that layer, and that print quality of arcs is poorer for those layers. In the attached file this occurs from layer 223-263 (Layer 222-223 jumps from 20-145 steps). Stepping through the layer gcode note lack of G2/G3 commands for the layers with slowed overhangs present.
[image]

### Expected results

Layers with overhangs should have arcs fitted, especially for portions of the layer not containing any overhang (but it should also still arc fit overhangs). Should see G3 commands for arcs of layers containing slowed overhangs.
[image]

### Project file & Debug log uploads

[arc fitting fail.zip](https://github.com/user-attachments/files/17690058/arc.fitting.fail.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

Related to:
https://github.com/SoftFever/OrcaSlicer/issues/3521
and possibly:
https://github.com/SoftFever/OrcaSlicer/issues/7315

malventano, 2024-11-10:
Digging into this further, there appear to be other useful pieces of [ArcWelder](https://github.com/FormerLurker/ArcWelderLib/releases/tag/1.2.0) that Bambu failed to implement in Studio, which may have carried forward into Orca:
- Allow travel arcs: This especially applies if 'Avoid crossing walls' is enabled and the model has curved surfaces. This results in travel arcs, which at the default high travel speed and acceleration, are very hard on the steppers (I've observed step loss due to this).
- Path tolerance is independent of sliced resolution: Currently the Resolution value is used for Arc Fitting. If the user increases this value in order to ensure arcs are fully fitted (default value results in segmented arcs on most models), then that increased value results in 'coarser' arc paths for those arcs which were not fitted. This also extends above to make the travel arcs have lower resolution, amplifying that issue as well.
- Allow 3D arcs: This would enable arcs to extend into Vase Mode.

Lastly, current versions of Orca crash while previewing gcode post processed by ArcWelder (when added as a post processor in 'others'), so there exists no simple workaround for users to achieve the above by using ArcWelder as a post processor within Orca.

CCS86, 2024-11-10:
Great summary!

Allowing 3D arcs would also allow continuity when using scarf seams.

Even if you add ArcWelder as a post processor, I found (a while back) that the post processing was only applied when you chose to export the gcode file, and NOT applied when choosing "print plate" or "send".

malventano, 2024-11-10:
> Even if you add ArcWelder as a post processor, I found (a while back) that the post processing was only applied when you chose to export the gcode file, and NOT applied when choosing "print plate" or "send".

Is that still the behavior? It would seem odd if the post processing was now called during the preview step only to have it revert back to the original gcode for the print. I'd confirm this but can't due to the [other bug](https://github.com/SoftFever/OrcaSlicer/issues/7433) :)

slc79, 2025-01-01:
Also having issues with this.. print quality in some situations is noticeably worse than from Bambu Studio because of it.

malventano, 2025-04-02:
Ping to not remain stale as this is still an issue.

gazi-faysal-jubayer, 2026-11-08:
/assign-me

gazi-faysal-jubayer, 2026-11-08:
A fix is up in #17369. Walls the overhang slowdown splits by speed now keep their arc fitting: each stretch at one speed is arc-fitted as the unsplit wall would be. Speeds and fan changes are unchanged.