OrcaSlicer/OrcaSlicer #7430: apply filename_format to file sended to printer

KubaMiszcz, 2024-11-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

now it is only works with export to gcode/3mf to disk
file sended to printer is always same
[image]

### Which printers will be beneficial to this feature?

Marlin

### Describe the solution you'd like

apply this filenameformat to alloptions - send/print/printplate etc

wahooper1, 2025-01-09:
I just realized this today as well. This would help tremendously with my print farm.

dudu631, 2025-04-02:
This would be really helpful!