OrcaSlicer/OrcaSlicer #7431: OrcaSlicer crashes on NixOS Sway (Wayland)

4e576rt8uh9ij9okp, 2024-11-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

NixOS 24.11.20241023.2768c7d (Vicuna) x86_64

### Additional system information

OS: NixOS 24.11.20241023.2768c7d (Vicuna) x86_64
Kernel: Linux 6.6.58
Shell: bash 5.2.32
WM: Sway (Wayland)
CPU: AMD Ryzen 9 9950X (32) @ 5.66 GHz
GPU 1: AMD Radeon RX 7900 XTX [Discrete]
Memory: 3.30 GiB / 30.44 GiB (11%)
Swap: Disabled
Locale: en_US.UTF-8

### Printer

Custom Manta M8P with CB1 + Mainsail

### How to reproduce

Open the slicer... wait 1-2s for it to crash.

### Actual results

It crashes with GTK critical errors. (reading all the bugs here it's a Xwayland/wayland issue)
Either Xwayland is bugged or it's a OrcaSlicer issue how it uses Xwayland.

### Expected results

Not crashing

### Project file & Debug log uploads

[warning]	2024-11-10 11:35:56.990437[Thread 0x00007f46287c1940]:Failed to load to hints.cereal. File does not exists. /home/user/.config/OrcaSlicer/user/hints.cereal
[error]	2024-11-10 11:35:57.316952[Thread 0x00007f46287c1940]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-11-10 11:35:58.154299[Thread 0x00007f46287c1940]:calc_exclude_triangles:Unable to create exclude triangles

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

I wish someone would be so nice and fix it as there are more and more Wayland Desktops available and people use them.

knosence, 2024-11-13:
I've witnessed the same on Wayland KDE and X11 KDE. I used unstable 2.2 which crashed and downgraded it to 2.11 which has, in like manner, crashed.

n00b0ss, 2024-11-26:
Maybe try adding a port number to your Host as suggested in this fedi post https://myside-yourside.net/@StarkRG/113526784879479099

InfiniteCoder01, 2025-02-12:
On my system it also seems to leak disk space until disk gets full, then it somehow cleans back up (after OrcaSlicer is killed)

closed 2025-05-21 by github-actions[bot]

InfiniteCoder01, 2026-02-12:
Still the case (it's interesting how I decided to try it exactly a year after last time). This is a wonderful slicer, but issues like this one prevent hundreds of people from using it, which is extremely sad.