OrcaSlicer/OrcaSlicer #7432: AMS Bug to change to Generic PLA

Ice-Cue, 2024-11-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Win10

### Printer

Bambu P1S mit AMS

### How to reproduce

Change Filament in THE AMS to Generic PLA
[drfgg.webm](https://github.com/user-attachments/assets/553d87db-e257-469b-a327-b52b1617163f)

### Actual results

it chaged not to PLA it will change to ABS

### Expected results

should setup to Generic PLA

### Project file & Debug log uploads

[drfgg.webm](https://github.com/user-attachments/assets/1d0b91ea-3136-4fa2-90e6-15a32ef5d913)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

dartrax, 2024-11-16:
I have exactly the same problem.

Duplicate of #7223 and #7292. But you have a very good Video showing the issue.

dartrax, 2025-02-15:
unstale

Ice-Cue, 2025-02-25:
yes. still not working on 2.3.0-beta2

closed 2025-03-17 by Ice-Cue

Ice-Cue, 2025-03-19:
> Thanks for sharing this!
> Could you please compare the old folder contents and the new folder
> contents and post the differences here? I think there must be something in
> the config that can be fixed by hand without reimporting, if we would just
> know what.

Is there a program what I can use for comparing the 2 folders?

dartrax, 2025-03-19:
I'd open those two folders im VS Code to compare them via the inbuild compare function!

dartrax, 2025-03-29:
Any success? ;-)

Ice-Cue, 2025-03-29:
the problem i cant do it. i deleted the old folger -_-