OrcaSlicer/OrcaSlicer #7434: Flush into object's supports can't be disabled for some objects when using by-object supports

Arakon, 2024-11-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Win 11

### Printer

Bambu A1

### How to reproduce

- Load the project file 
- Slice as set
- Notice how the gun-sights-3mm-led.stl and front-1.stl have PETG interface material flushed into the supports
- Setting or unsetting the option both globally and by-object makes no difference
- Notice how frame-40mm-bearing-rail-1.stl does NOT have PETG flushed into the  supports

### Actual results

Some objects still get PETG/secondary color flushed into supports, which will lead to the supports falling apart mid-print
[image]

### Expected results

When flush into object's supports is disabled, it should be disabled for all objects

### Project file & Debug log uploads

[frame-petgsupports.zip](https://github.com/user-attachments/files/17691607/frame-petgsupports.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

closed 2025-02-16 by github-actions[bot]

dstulken, 2025-04-03:
I am also seeing this issue, in one of the 2.3.0 Betas.   Were you able to find a solution?

Arakon, 2025-04-03:
No, unfortunately. Issue still happens in 2.3.0 final, too.