OrcaSlicer/OrcaSlicer #7435: Getting over extrusion leading into front overhang on Benchy and under extrusion when moving away, when enabling "slow down for curling perimeter"

ImDisaster, 2024-11-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11 Home, 23H2

### Printer

Sovol SV08

### How to reproduce

Turn on "slow down for curled perimeters" to create issue.  Turn off to fix.

### Actual results

About 4mm up from bottom to 12mm from bottom it over extrudes entering front corner of ship and under extrudes leaving the corner.  Effect starts about 5mm from corner.  

### Expected results

[image]

### Project file & Debug log uploads

[
[image]
[3dbenchy_fixed_hopefully.zip](https://github.com/user-attachments/files/17692061/3dbenchy_fixed_hopefully.zip)
](url)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

Added file project file with slow for curled perimeters off.  Turn on to create issue.
[3dbenchy_fixed_hopefully.zip](https://github.com/user-attachments/files/17692054/3dbenchy_fixed_hopefully.zip)
[image]

igiannakas, 2024-11-11:
Enable extrusion rate smoothing. Your extruder cannot change its filament position fast enough causing it to extruder more when rapid speed changes happen. 

Also calibrate your pressure advance - this should mitigate the issue to a large extent

ImDisaster, 2024-11-12:
I ran all the calibrations in Orcaslicer slicer including pressure advance and retraction.  I have tried with the stock settings in Orcaslicer and with the modified settings.  I have had the issue from .02-.04 smoothing.  Default setting was .035.  .02 smoothing has shown the best results overall.  I have tuned retraction as well...but it happens on the stock retraction settings too, which were .5 mm and 30 mm/s.

I also saw the same issue with the Qidi Plus4.  Turning off the slow for curling fixed that one too.  It is strange it only affects a specific range of the Benchy and an overhang test (50-70).  It is almost as if the overhang and curling have an additive effect which creates the issue.

igiannakas, 2024-11-12:
I’m confused. You mention smoothing - is that the PA smooth time you are referring to? 

Retraction has nothing to do with this artefact. This is purely the extruder not being able to keep up with the rapid speed change when going from fast to slow print speeds. 

Solutions include:
1. Reduce external perimeter and internal perimeter accelerations - higher accelerations cause more stress on the extruder when rapidly changing speeds
2. PA value is not spot on - the tests may be run but real world is showing artefacts so tuning the PA value (not smooth time but PA) is improtant
3. Enable extrusion rate smoothing and set this to 150 with segment length of 1 or 2. This should smooth out the changes in speed on the slicer side if the printer cannot do this well on its own.

you can disable the slowdown for curled perimeters feature but you’re not solving the problem, which is the printer not being able to respond well to rapid speed changes. It’s best to address the cause of this rather a feature that relies on the printer doing what it’s told well, if that makes sense?

igiannakas, 2024-11-12:
For context the below with curled perimeters enabled on a calibrated 2.4 Voron, so it can be done :)

[image]

igiannakas, 2024-11-12:
The extrusion rate smoothing on the wiki explains the above pretty well. Have a read ;)

boehser-enkel, 2025-01-24:
Will try out this option because I get even worse surfaces. IMO it should not be needed to mitigate problems by a feature silently activated through an update.

igiannakas, 2025-01-24:
Seeing variations in extrusion consistency when moving from fast to slow to fast extrusions is the core artefact of incorrectly calibrated Pa. I’d suggest to focus on getting it calibrated right for your external perimeter speeds and accelerations and it should improve or resolve completely.

If it doesn’t, a do over the extruder is highly recommended. 

If nothing works the software mitigation can help but it really should be a last resort in my opinion.

boehser-enkel, 2025-01-24:
Improved but not good. I rather miss this option as I’m also not having problems with curled overhangs. Then again why is this activated by default?

igiannakas, 2025-01-25:
Because in general it gives better results as more time is spent cooling areas that are prone to curling upwards. 

Prusa also has this option enabled by default for the same reason (and you can’t turn it off).

gscsonda, 2025-03-22:
This issue is very visible with slow speed filaments like TPU. Here my findings
https://www.reddit.com/r/3Dprinting/comments/1jh2rk6/orca_23_still_has_slow_down_for_curled_perimeters/

igiannakas, 2025-03-22:
Extrusion consistency and volume is not affected because of this feature. This feature highlights extrusion consistency issues that exist on the printer, if that makes sense.

If it doesn’t work for you, disable it. But personally I’d invest time in fixing what’s causing your printer to extrude more material at slower speeds :)

gscsonda, 2025-03-22:
If a user has to go this down the rabbit hole to fix the result of an option, for me it means the option shouldn't even be on by default to start. Its easier to always discredit user and blame calibration.

Sure going in a constant speed, then accelerating for a fraction of a second to suddenly slow down (slicer pic on Reddit) all in a length of 1cm is perfectly fine and presents no failure point. Right? Why does it accelerates before slowing down?

Corexy with absurd accelerations are becoming the norm. Why dont we try to make it more plug and play possible?

igiannakas, 2025-03-22:
I’m not intending to discredit anyone - or blame calibration. However extrusion needs to be consistent irrespective of speed to achieve good results, not only for this feature but throughout.

 If the calibration is off you’ll encounter similar issues in other places in the print - eg after overhangs, after bridges and when changing speed due to layer time. 

For info, prusa slicer has this option enabled by default and cannot be disabled - the feature was ported from prusa slicer last year and works in an identical manner.

The below was taken with  a Voron:
[image]

If the extruder is struggling to manage the filament, the user should enable extrusion rate smoothing to allow a more gradual transition. 

This would benefit the print throughput - bridges, overhangs and curled perimeters.

gscsonda, 2025-03-22:
Ok. I'll take your advice and play around with rate smoothing. What im failing to understand is why the acceleration before slowing down and the increase of flow also right before

[image]

igiannakas, 2025-03-22:
> Ok. I'll take your advice and play around with rate smoothing. What im failing to understand is why the acceleration before slowing down and the increase of flow also right before
> 
> [image]

Post your project file - let me take a look.

gscsonda, 2025-03-22:
[3DBenchy.zip](https://github.com/user-attachments/files/19405623/3DBenchy.zip)

There you go.
Yeah, so adding 300mm3/s2 as per the recommendation on the tooltip of Extrusion Rate Smoothing mitigated a bit the issue without completely resolving it. I'm sure I could fine-tune the value better, but I suppose I'm not advanced enough to fully understand the wiki.

What I've found on the other hand is that there's another default setting conflicting, which is overhang speed 25%,50% at 40mm/s.
40mm/s being faster than the regular outer wall without overhang. Setting at 30mm/s smoothed things out, mitigated even more the curling, but was still persistent (almost not perceptible). but i'm a perfectionist so ... haha

My camera is pretty bad so it's almost impossible to see.

[image]

My take from all this is that if my cooling is enough and I see no curling up during perimeters like this, I should disable the feature. Am I wrong?

igiannakas, 2025-03-22:
I’m using the below speed settings for mine:
Outer wall: 140mm/sec
Overhang 10-25: 0
25-50: 80mm/sec
50-75: 60mm/sec
75-100: 40mm/sec

Bridge speed is set to 40mm/sec too

You don’t need to go crazy slow for overhangs - below 30-40mm/sec I’ve seen very little benefit.

gscsonda, 2025-03-22:
I'm using TPU.

If I slice with the recommended settings, the flow and speed increase just before the slowdown down for curled perimeter is even more accentuaded, nullifying the slowdown altogheter

[image]

igiannakas, 2025-03-23:
This looks like a bug as it seems like it’s not respecting the flow rate of the filament.
I’ll take a look but I need a project file attached

gscsonda, 2025-03-23:
Ive added a zip on a previous post with my 3mf inside.
Cutting the benchy and isolating the hull accentuates the bug. Thanks for looking into it.

linouxis9, 2025-04-04:
I'm able to reproduce the same issue as you with a Qidi Plus 4 @gscsonda!

igiannakas, 2025-04-04:
I think this Pr here fixes it. https://github.com/SoftFever/OrcaSlicer/pull/9210
Please test the latest nightly tomorrow morning once it is produced.

igiannakas, 2025-04-04:
Ok the PR above was not correcting for this. I've raised a new one here: https://github.com/SoftFever/OrcaSlicer/pull/9215

PR #9215 (2025-04-05): Fix speed up during overhang slowdown when curled perimeters is enabled.
On some edge cases mostly observed when curled perimeter detection is enabled, and the perimeter speeds are limited due to flow rate limits, the resulting overhang speed calculation creates a speed ramp up. This could also happen with the curled perimeter detection being disabled too.

This happens because the calculated overhang speed is not limited by the actual print speed but instead calculated by the speed thresholds in the overhang % boxes. Typically this is not an issue, as the overhang speed should be set to less  [... cut]
closed 2025-04-05 by Noisyfox