OrcaSlicer/OrcaSlicer #7437: Diferrent and weird gcode after saving and reopening project

szymswiat, 2024-11-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Ubuntu 24.04

### Printer

Bambu Lab A1

### How to reproduce

1. Create new project and add [attached](https://github.com/user-attachments/files/17692439/test_cyl_thread.stl.zip) object to a plate
2. I set preset to 0.08mm extra fine however the issue occured with other presets as well
3. Slice plate and observe the result

[image]
[image]

5. Save project and close Orca Slicer
6. Open it and slice again

### Actual results

After reopening a project and slicing a plate Orca Slicer adds weird bridge and top surface layers.

[image]

### Expected results

Unwanted bridge in the air should not appear.

### Project file & Debug log uploads

[test_cyl_thread2.3mf.zip](https://github.com/user-attachments/files/17692467/test_cyl_thread2.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Arakon, 2024-11-10:
The STL is not manifold. Is it possible you repaired it the first time, but the repair gets lost when reloading the project?
[test_cyl_thread-repaired.zip](https://github.com/user-attachments/files/17692922/test_cyl_thread-repaired.zip)
Can you try the same thing with this repaired version of the stl?

szymswiat, 2024-11-18:
Repaired? According to the pop-up notification, repair is not supported under Linux.

closed 2024-12-27 by szymswiat