OrcaSlicer/OrcaSlicer #7447: To make available the estimate printing duration on placeholders

antonini, 2024-11-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I'm tring to add to the output gcode a parameter containing the estimate duration of printing in seconds. Other slicer softwares do that creating an output of print_time. 

### Which printers will be beneficial to this feature?

Klipper, Marlin

### Describe the solution you'd like

Add the variable so it can be used by developers and even on the default gcode so fluid can display the estimate time on the interface

Venuman22, 2024-12-29:
Have you tried adding {print_time} to your Filename Format section on orca slicer? I think it has to be all lowercase for it to work.

antonini, 2024-12-30:
Yes. That placeholder is not available. I tried on triggers "machine start", "machine end", "before layer change", "after layer change".

I was looking on the source code and haven't found anything about that placeholder.