OrcaSlicer/OrcaSlicer #7451: Appimage - error while loading shared libraries: libwebkit2gtk-4.0.so.37

TheDevMinerTV, 2024-11-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Ubuntu 24.10

### Additional system information

AMD Ryzen 5 7640U
32GB RAM

### Printer

Anycubic i3 Mega

### How to reproduce

1. Run the AppImage on Ubuntu 24.10

### Actual results

It fails with `/tmp/.mount_OrcaSl2hqCOw/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory`

### Expected results

OrcaSlicer should start like normal.

### Project file & Debug log uploads

Not Applicable

### Checklist of files to include

- [ ] Log file
- [ ] Project file

nils-s, 2024-11-16:
Reproduced on Fedora 41 / KDE Spin / x86-64 (Kernel 6.11.7), fully-updated as of Nov. 16th

Fixed by installing `webkit2gtk4.0` and `webkit2gtk4.0-devel` packages (don't know if one of them would have been enough, did both in one go, after that OrcaSlicer is working again).

Snippet to C&P for the fix (assuming you're on Fedora or similar):

```bash
sudo dnf install webkit2gtk4.0 webkit2gtk4.0-devel
```

joseviniciusnunes, 2024-12-04:
same error on Ubuntu 22

powermad80, 2024-12-28:
> Reproduced on Fedora 41 / KDE Spin / x86-64 (Kernel 6.11.7), fully-updated as of Nov. 16th
> 
> Fixed by installing `webkit2gtk4.0` and `webkit2gtk4.0-devel` packages (don't know if one of them would have been enough, did both in one go, after that OrcaSlicer is working again).
> 
> Snippet to C&P for the fix (assuming you're on Fedora or similar):
> 
> ```shell
> sudo dnf install webkit2gtk4.0 webkit2gtk4.0-devel
> ```

dnf returns that these packages are already installed. Error continues to occur on attempting to launch AppImage on Fedora 41 KDE.

Senectus, 2025-01-12:
Same issue here, Brand New Fedora 41 install, Nvidia and intel.

`

senectus@fedora:~/AppImages$ ./OrcaSlicer_Linux_Ubuntu2404_V2.2.0.AppImage 
[2025-01-12 19:00:57.319658] [0x00007fc10e889740] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.752: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.753: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.754: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.755: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.755: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.756: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.757: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.758: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.759: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.759: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.900: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.903: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.903: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.903: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.903: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.904: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.904: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.904: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.904: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.909: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.909: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.909: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.987: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.987: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.987: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.987: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.994: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.994: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.994: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:00:57.994: gtk_widget_set_size_request: assertion 'width >= -1' failed
[... 171 lines cut ...]
(orca-slicer:12784): Gtk-CRITICAL **: 19:00:58.619: gtk_window_resize: assertion 'height > 0' failed
Failed to create GBM buffer of size 1200x713: Invalid argument
No such file or directory

(orca-slicer:12784): Gtk-CRITICAL **: 19:01:00.037: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:01:00.037: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:01:00.039: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:01:00.039: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:01:00.040: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:01:00.041: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:12784): Gtk-CRITICAL **: 19:01:00.041: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (orca-slicer:12784): CRITICAL **: 19:01:00.056: Cannot register URI scheme wxfs more than once

** (orca-slicer:12784): CRITICAL **: 19:01:00.056: Cannot register URI scheme memory more than once
Failed to create GBM buffer of size 818x624: Invalid argument
`
[image]
The app just seems to hang on this

nils-s, 2025-01-13:
> Same issue here, Brand New Fedora 41 install, Nvidia and intel.
> 
> `
> 
> senectus@fedora:~/AppImages$ ./OrcaSlicer_Linux_Ubuntu2404_V2.2.0.AppImage [2025-01-12 19:00:57.319658] [0x00007fc10e889740] [trace] Initializing StaticPrintConfigs add font of HarmonyOS_Sans_SC_Bold returns 1 add font of HarmonyOS_Sans_SC_Regular returns 1 add font of NanumGothic-Regular returns 1 add font of NanumGothic-Bold returns 1
> 
> [... snip ...]
>
> The app just seems to hang on this

That's not actually the same issue, though. The OP had an error regarding a shared library (libwebkit2gtk), which was explicitly mentioned in the output when trying to start the application (i.e. the relevant GTK library was missing, and the app could not start at all).

Your issue is something else: the library seems to be present, the application is starting, but then something fails. Not sure what the cause is, but the solution that worked for me (installing the missing lib) will almost certainly not work for you. Maybe your issue should be moved to its own, separate issue...

Senectus, 2025-01-14:
because  @nils-s , that was the instructions.
`Orca Slicer for Linux (newer-distros)
Version – 2.2
Minimum Requirement – Ubuntu 24.04, Fedora 40, Debian 12
File Size – 94.2 MB
Last Update – Nov 10, 2024`

The DOWNLOAD button next to that links to:
https://github.com/SoftFever/OrcaSlicer/releases/download/v2.2.0/OrcaSlicer_Linux_Ubuntu2404_V2.2.0.AppImage

I figured the naming convention was just an artifact of ubuntu being very popular.

I'm on Fedora 41 now (was on Bazzite but decided immutable wasnt my cup of tea)
Orca "installs" (as much as an appimage can) and runs but I get no graphical view of the print object.

kandrejevs2, 2025-01-16:
```
./OrcaSlicer_Linux_Ubuntu2404_V2.2.0.AppImage 
[2025-01-16 20:07:08.538386] [0x00007295e5764480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
loading network module at /home/kriss/.config/OrcaSlicer/plugins/libbambu_networking.so
after dlopen, network_module is 0x64ee68455790

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.162: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.162: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.162: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.163: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.163: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.163: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.165: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.166: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.168: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.169: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.171: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.172: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.174: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.175: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.273: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.278: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.278: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.278: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.278: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.280: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.280: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.280: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.280: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.288: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.289: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.289: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.416: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.416: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.429: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:09.429: gtk_widget_set_size_request: assertion 'width >= -1' failed

[... 189 lines cut ...]
(orca-slicer:11685): Gtk-CRITICAL **: 20:07:11.854: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:11.854: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:11.857: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:11.869: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:11.869: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:11.876: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:11.877: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:11.877: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:11.880: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:11.880: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11685): Gtk-CRITICAL **: 20:07:11.882: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Segmentation fault (core dumped)
```

it crashes when printer has mainsail web ui, editing `~/.config/OrcaSlicer/user/xxxxx/machine/printer.json ` and removing IP from `print_host_webui` and leaving just moonraker seems to stop crashing, it might be related to something with sandboxed browser, had similar issues with IDE in the past

AiOGajo, 2025-01-21:
Elementary OS 7.1:

I try to execute: 
`./OrcaSlicer_Linux_Ubuntu2404_V2.2.0.AppImage`

I get the error: 
` /tmp/.mount_OrcaSlOnaBUu/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.1.so.0: cannot open shared object file: No such file or directory`

VO1XA, 2025-02-06:
> Elementary OS 7.1:
> 
> I try to execute: `./OrcaSlicer_Linux_Ubuntu2404_V2.2.0.AppImage`
> 
> I get the error: ` /tmp/.mount_OrcaSlOnaBUu/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.1.so.0: cannot open shared object file: No such file or directory`

I'm running Kubuntu 24.10 and have the exact same problem. Have you had any luck?

atazx, 2025-02-06:
> I'm running Kubuntu 24.10 and have the exact same problem. Have you had any luck?

Hi, on Kubuntu 24.10 you just need to install libwebkit2gtk-4.1-0
`sudo apt-get install libwebkit2gtk-4.1-0`

VO1XA, 2025-02-06:
Didn't work for me. That package is already installed

Zikoel, 2025-02-18:
Same problem here on Ubuntu 24.04, orca ask for `libwebkit2gtk-4.0.so.37` the main ubuntu repository suggest me only the `libwebkit2gtk-4.1-0` but whit it the problem remain.

closed 2025-06-05 by github-actions[bot]