OrcaSlicer/OrcaSlicer #7454: Gradual Infill Steps

TheAtomicSoul, 2024-11-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Gradual Infill Steps: Both to infill and support!
This is the ONLY FEATURE I MISS from Cura: It both helps REDUCE WEIGHT and REDUCE MATERIAL USED. 

Allows better support of top surfaces where needed with MUCH less material use overall.

[image]
[image]
[image]

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add Gradual Infill Step options from Cura to both infill and support
- Gradual Infill Steps
- Gradual Infill Step Height

### Additional context

Previously submitted and closed by a bot:
- by me: https://github.com/SoftFever/OrcaSlicer/issues/3100
- by someone else: https://github.com/SoftFever/OrcaSlicer/issues/5083

TheAtomicSoul, 2024-11-22:
Bump to keep from getting stale.

tlhintoq, 2024-12-08:
Came to log the same request and it was already here.

Dear gawd yes... Would love to have an 80% infil that halves, halves, halves... So it starts at nearly nothing then becomes this nice density just below the external top surface layers.
I got nowhere with some of the infil patterns and was starting to plan staggered layers of modifiers like layers in a cake to achieve the same - but what a mess to set up especially on several things.

Nandox7, 2024-12-28:
Creality added to it's later slier version (5.1) a feature called AI Infill that is someone what is asked here.
In summary more infill density near the walls ad less away from the walls.

https://www.youtube.com/watch?v=yUhf6j6wNY8

tlhintoq, 2024-12-30:
That would be freaking awesome!

TheAtomicSoul, 2025-02-07:
Bumping this to keep it alive. REALLY wish this would be implemented.

coltography, 2025-02-07:
Was looking into this after discovering adaptive cubic and surprised its not a feature in orcaslicer already! Hope it can be added some day.

hotellonely, 2025-03-14:
bumping again we REALLY need this instead of the poor adaptive cubic which is really poorly designed.

tlhintoq, 2025-03-14:
**_Bandaid_**:  Use modifiers.

[image]

If you want to get really crazy you can even change pattern with a modifier.
From 5% cubic to 15% hex if you want.

[image]

Nandox7, 2025-03-18:
That's cool and will allow on a per layer change, but does not cover the need to have it change horizontally.
I imagine with even more work you can do several modifier blocks that get it to change the infill depending if it's closer to the all or more at the center, but it would be mad work and depends on how complex is the part.

tlhintoq, 2025-03-19:
To be fair the Cura feature this would be cloning doesn't do a smooth transition either.
It doesn't change from 1% to 3% to 5%.
It does an increase of halves (doubling) for the entire part at the prescribed intervals.

hotellonely, 2025-03-19:
> To be fair the Cura feature this would be cloning doesn't do a smooth transition either. It doesn't change from 1% to 3% to 5%. It does an increase of halves (doubling) for the entire part at the prescribed intervals.

Yes, but it's quite effective. In fact you don't need a 80% for perfect pillow free experience, in my real world tests usually 25% is already pretty much pillow free.

ariefadha, 2025-07-08:
I really need this feature, hope someone will add this, thanks.

TheAtomicSoul, 2025-08-11:
Adding another bump. I don't get how people don't see how useful this would be.

TheAtomicSoul, 2025-10-22:
Ultimaker Cura Gradual Infill Steps https://youtu.be/z038SdtIk0c I used this ALL THE TIME in Cura and it's my BIGGEST MISSED FEATURE.

Creality slicer(5.1)  AI Infill: https://youtu.be/yUhf6j6wNY8 (not really a good visual example, but surprisingly there isn't one?)

inspectionsbybob, 2026-08-28:
Bump to keep this active.
This is the only reason I have 4-5 different slicers on my system.  Each one of them has 1-2 features that no one else has implemented, going all the way back to slic3D

There has been lots of work done on this so I am confused as to why no-one has imported it.

PLEASE, PLEASE, PLEASE

TheAtomicSoul, 2026-10-15:
Bumping AGAIN because I REAAALY want this feature added. It's the ONLY feature I miss from another program. 

Also: I got a message about this being related to this post: https://github.com/OrcaSlicer/OrcaSlicer/pull/15343