OrcaSlicer/OrcaSlicer #7461: Login and Bambu Cloud Sync Causes Hanging / Crash

x24a298, 2024-11-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.20

### Operating System (OS)

macOS

### OS Version

macOS Sonoma 14.5

### Additional system information

MacBook Air M2 2022 (8gb)

### Printer

Bambu Lab P1S; imported 0.4mm Nozzle on initial launch

### How to reproduce

1. Log out or open fresh install
2. Home -> Login or register -> log in with Apple* -> Safari redirect* -> Touch ID login*
3. Redirect back to OrcaSlicer from Safari
4. Ignoring error popup which cannot be interacted with, select "Yes", "No", or close the dialogue box when prompted to sync personal data from Bambu Cloud

*Did not test alternatives

### Actual results

1. Error popup after being directed back to OrcaSlicer after Safari Login: this can not be interacted with:
[image]

2. After moving the Sync window out of the way, responding either "Yes", "No", or closing the dialogue box leads to the app almost immediately going unresponsive (+ beach ball of death), requiring a Force Quit and leading to Apple bug report popup:

[image]

### Expected results

Orca should sync with Bambu Cloud. If there are corrupt files that cannot otherwise be synced, ideally there would be some kind of error handling, especially when selecting "No" on the Sync prompt.

### Project file & Debug log uploads

Project File: [Login_Bug.3mf.zip](https://github.com/user-attachments/files/17726690/Login_Bug.3mf.zip)
Debug Log: No debug log file generated :(

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

Assuming this is an issue with the files in Bambu Cloud, I am happy to share those (if it is possible) to help remedy the issue. I have not had this issue when running Bambu handy and using Sync, but have not perfectly replicated conditions.

jamesc83, 2024-11-28:
Same bug here, Is there a fix?

theosib, 2024-12-31:
The latest 2.3.0 nightly still has this problem.

theosib, 2025-01-01:
This seems like a pretty serious bug. Do you think it would be a good idea to have a developer mark it as a blocker for the next release?

closed 2025-04-10 by github-actions[bot]