OrcaSlicer/OrcaSlicer #7462: Inherits tag broken if the parent config is in the

donmahallem, 2024-11-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

Not relevant

### Printer

Any

### How to reproduce

1. Create two filament profiles in `user/default/filament` and inherit from one from the other one with `{"inherits": "other_profile_filename"}`. The second fill with {"inherits": "Voron Generic PLA"}
2. Open OrcaSlicer

### Actual results

Only the second profile will be visible and an error is thrown in the log files

### Expected results

It should inherit the other profile file as expected

### Project file & Debug log uploads

[error]	2024-11-13 07:18:55.678076[Thread 0x0000460c]:can not find parent for config ...\user\default\filament\other_profile_name.json!

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

This is probably related to this line https://github.com/SoftFever/OrcaSlicer/blob/b7bba13789a971bd5e594896e7c8d9f616bd7c4a/src/libslic3r/Preset.cpp#L1143-L1147
from my limited understanding of the code base OrcaSlicer loads every preset file and checks if the inherited preset is known before knowing all presets in the folder and thus only being able to reference previously known profiles(builtin profiles). 

While I know with inheritance loop issues could become an issue it would be great if the inheritet preset loading could become recursive and make it possible to use intermediate profiles instead of copying always the same settings to new files.

johyphenel, 2025-01-06:
+1, it would be really nice to be able to set inheritance between user presets instead of being forced to inherit from system presets

closed 2025-04-14 by github-actions[bot]

johyphenel, 2025-05-04:
Filed new FR since this got closed: https://github.com/SoftFever/OrcaSlicer/issues/9536