OrcaSlicer/OrcaSlicer #7464: Jerk settings not suitable for default Voron's when run with SnT

oernster, 2024-11-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Not exactly.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

I would like all jerk settings set to 5 as a default since Shake n Tune (a tool for input shaping on Vorons and klipper printers) works much better to obtain good results with 5 instead of your defaults which I presume you inherited from a Bambu printer.
Naturally people can fine tune to their needs but I feel that 5 is a good starter for 10 for all jerk.

recrudesce, 2024-11-13:
Jerk settings in your slicer are not used by Shake n Tune, which is a separate tool, and uses the jerk settings you've defined in your printer config within Klipper.

Orca settings are either set to "use whatever the printer config says to use" (ie is set to 0) or is used to hard set the value if required and is only set in the sliced gcode for an object you have sliced.  SnT doesn't use any sliced objects, nor does it use a slicer.

Confused where you've made the association.

oernster, 2024-11-13:
Jerk settings as defaults in Orca are not suitable for most Voron users.  For most users, jerk of 5 for everything is much more reasonable.  I completely get that scv is set in klipper in your printer.cfg but your slicer settings will be used in priority if you forget to set this value (not a remembered setting by many, though I have it set, peronally).

SnT might have been a bit of an irrelevant reference here, I understand, though as a benchmark it is recommended by the Voron community to use scv of 5.

What I'm after is a default Voron profile which sets all the jerks to 5.  Is this not possible?  If advanced users want to fine tune this then they certainly can; I just feel the default values set are inappropriate for a Voron 2.4 or Trident printer.

ukutghost, 2024-11-13:
Its completely possible, you could do it. Just need to update the relevant JSON files and PR them.... but why are the current values not suitable? Have there been cases where they have created a problem?

recrudesce, 2024-11-13:
My V2.4 runs absolutely fine with the jerk settings set in the default Orca profiles, as have all the other Vorons I've built for other people.  I have also had reports of great success when advising people to use the default profiles for their Voron printers.

The defaults seem to work fine.  Can you give references to where "it is recommended by the Voron community to use scv of 5." ?

But yeah, raise a PR if you think it needs changing, then that will be reviewed as per standard development cycles.

recrudesce, 2024-11-13:
SuperSlicer sets the following jerk settings for all Vorons (it just uses whatever is configured in the Slic3r profiles on their github):
```
machine_max_jerk_e = 2.5
machine_max_jerk_x = 10
machine_max_jerk_y = 10
machine_max_jerk_z = 0.2
```
https://github.com/slic3r/slic3r-profiles/blob/ca25c7ec55dcc6073da61e39692c321cdb6497dc/Voron.ini#L219C1-L222C25

PrusaSlicer sets them identically:
https://github.com/prusa3d/PrusaSlicer/blob/5dc04b4e8f14f65bbcc5377d62cad3e86c2aea36/resources/profiles/Voron.ini#L222

Orca sets them as per here and is common across all the Vorons, regardless of whether it's a 2.4, 0.1 etc:
https://github.com/SoftFever/OrcaSlicer/blob/b7bba13789a971bd5e594896e7c8d9f616bd7c4a/resources/profiles/Voron/process/fdm_process_voron_common.json#L108

Therefore NO slicer sets it to the "Voron Community recommended SCV of 5".  Klipper has a default of 5, that's just cos Klipper is printer agnostic, therefore they set it low so you don't blow up your cheap Anet or something :P  5 is a safe starting point to tune from, not the expected max of the printer.

oernster, 2024-11-13:
5 is the typical benchmark for SnT and I assumed, perhaps incorrectly, that you want this for printing as well as tuning.

https://github.com/slic3r/slic3r-profiles/blob/ca25c7ec55dcc6073da61e39692c321cdb6497dc/Voron.ini#L219C1-L222C25

https://github.com/prusa3d/PrusaSlicer/blob/5dc04b4e8f14f65bbcc5377d62cad3e86c2aea36/resources/profiles/Voron.ini#L222

https://github.com/SoftFever/OrcaSlicer/blob/b7bba13789a971bd5e594896e7c8d9f616bd7c4a/resources/profiles/Voron/process/fdm_process_voron_common.json#L108

recrudesce, 2024-11-13:
SnT uses whatever SCV you have configured in your Klipper config

https://github.com/Frix-x/klippain-shaketune/blob/66f5e32e4c8930afab278fcd898b47c227cc464c/shaketune/commands/axes_shaper_calibration.py#L33

You can override this if you want, but as the default Voron Klipper config has SCV set to 5 (conservative default config) then SnT will use an SCV of 5.  If you had it set to 343209432 in your Klipper config, then SnT would use that value instead.

oernster, 2024-11-13:
The experts who hang out in resonance testing and unicorn chasing in the Voron discord recommend 5 scv for SnT which is why I was so focused on that number originally; nothing to do with klipper default setups.
However, it seems to me that there is a lot of resistance to this potential change so let's just drop the issue/ticket.

https://github.com/Frix-x/klippain-shaketune/blob/66f5e32e4c8930afab278fcd898b47c227cc464c/shaketune/commands/axes_shaper_calibration.py#L33

oernster, 2024-11-13:
> No resistance at all, just trying to understand. Happy to be proven wrong with valid data to back the suggestion up.
>
> However, feel free to raise a PR with the changes and make sure you include detailed explanation as to why it needs changing so SoftFever can understand why it needs changing :)
>
> -

Thanks for your understanding.  However, I feel that I do not have a strong enough case for this change and only have e.g. an expert like Reth to go by who recommends scv of 5 for your SnT tuning; where scv closely corresponds to jerk.  I had not considered that print jerk values (or scv for that matter) could potentially be wildly different from your tuning settings and you still may get reasonable print quality.

ukutghost, 2024-11-13:
5 is just the default for Klipper and a safe default at that. SnT will just use what you put in your klipper config, if you put anything at all, and your slicer is capable of overriding that, which they do in many modern slicer profiles. SCV is also a fairly complicated thing in itself and ideally needs tuning based on the features and their accelerations, and how you want your printer to behave.

To say its not suitable without evidence beyond hearsay and an example of a single benchmark which takes the default or a pre-set value seems a bit extreme and ignores the tuning goal of whoever set the default values originally. The fact SnT takes it as a variable from the printer config suggests its a value which they expect will vary from machine to machine and change with the tuning goal. 

But as I said, there's nothing stopping you from updating the relevant JSON files, and submitting them as a PR to address this issue if you feel it needs doing.

oernster, 2024-11-13:
I have insufficient evidence to support my case so I will be dropping the issue/ticket.

recrudesce, 2024-11-13:
Don't forget to close it :)

closed 2024-11-13 by oernster