OrcaSlicer/OrcaSlicer #7471: Cannot Select Printer when Creating Custom Filament

LevonBecker, 2024-11-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

15.1

### Additional system information

Mac Studio
CPU: Apple M2 Max (12 core)
MEM: 64GB
HD: 2TB

### Printer

Creality K1 Max + Bambu A1

### How to reproduce

1. Prepare Tab
2. Select Filament Dropdown box
3. Select Add/Remove Filament
4. Select Custom Filaments
5. Select Create New
6. Select Vendor, Type, Serial, Filament Preset create based on current filament `Creality Generic PLA`
7. Creality printer not listed to select... and then create

### Actual results

[image]

### Expected results

Creality printer in list from printer profiles to select and then create. 

### Project file & Debug log uploads

[debug_Wed_Nov_13_09_14_32_31581.log.0.zip](https://github.com/user-attachments/files/17737190/debug_Wed_Nov_13_09_14_32_31581.log.0.zip)
[laser_bracket.3mf.zip](https://github.com/user-attachments/files/17737198/laser_bracket.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

I am logged into Bambu Labs and have the discovery plugin installed. Maybe that makes it a little janky with a non Bambu Labs printer in printer profile? Thanks for the help!

LevonBecker, 2024-11-13:
Oh even if I select the Bambu A1 printer it gives me an error. When I then select `Create`
[image]

zeemyself, 2024-11-15:
[image]

Dark mode UI is blend in with checkbox button

Ahtran360, 2024-12-04:
[image]
My printer doesn't even show up as an option. None of the printers show up.

MacYver, 2024-12-04:
I can report the same Problem with Orca Slicer 2.2.0 with Windows 11.
I have a Flashforge Adventurer 5M and my Screen looks similar to @Ahtran360 Screen, no Printer that can be selected.

ajcerezo, 2024-12-05:
I have the same problem with Orca Slicer 2.2.0 with linux.

karaktaka, 2024-12-07:
I'm also affected by this:

Windows 11 - OrcaSlicer 2.2.0

[image]

mikeduddy, 2024-12-12:
Just jumping in to add that this issue also exists on the MacOS ARM64 version in 2.1.0 and 2.2.0. It's a REALLY annoying bug...

karaktaka, 2024-12-14:
There is actually an open pr that fixes the issue. I checked it out, compiled and tested. Works for me. https://github.com/SoftFever/OrcaSlicer/pull/7477

ajcerezo, 2024-12-16:
My problem is solved with this solution,  #7477 

Thank you!