OrcaSlicer/OrcaSlicer #7474: Supports for Large Flat Areas Using Tree Supports Creates mid-air geometry

saeflizard, 2024-11-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

V2.2.0 Official Release

### Printer

Voron 2.4

### How to reproduce

1) Open model with large, flat overhangs:
[image]

2) Filament selected is 0.4mm PC with default settings plus clicking default PA and temp adjustments
[image]

3) 60% gyroid, 5 walls, 5 top+bottom, default speeds, arc fitting enabled, +0.1mm elephant foot comp, 3 skirts combined at 1 layer, auto brim, 

4) support settings: Auto Tree Supports with both defualt and organic supports

5) slice

### Actual results

Entire print looks good at first
[image]

go to layer 10 and issue appears
[image]
[image]
[image]
[image]
[image]

Circled object shows up dis-attached to other support on layer 5, causing issues
[image]

### Expected results

Support generation should not create floating sections that are not created without support beneath.

### Project file & Debug log uploads

[e3d-single-tray-lm8uu.zip](https://github.com/user-attachments/files/17739331/e3d-single-tray-lm8uu.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

Simple fix is to switch support type, it would not be a good day for someone if this happened on a larger print

khoanguyen-3fc, 2025-02-12:
I often face this issue too. Sometimes, changing the support type for each object works, but other times it doesn’t.

For example, I need tree support (since the trees reach, so support doesn’t start on the object, making removal easier), but the object also has a large flat area at a lower height, which would cause this bug.

khoanguyen-3fc, 2025-03-06:
@saeflizard I just realized that the Hybrid Tree style resolves my issue, and your attached model, too.

The original problem with the default tree style was that too many overhanging nodes merged back to the build plate, and I don't think it's easy to fix this using just the vanilla tree style.

[image]

closed 2025-06-12 by github-actions[bot]