OrcaSlicer/OrcaSlicer #7478: Text tool crashing with extreme values and use surface

Berkays, 2024-11-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

14.7.1

### Printer

P1S 0.4

### How to reproduce

1) Add a text to any object surface
2) Enable Use Surface
3) Adjust boldness or any other parameter to extreme values such as +-90 or higher.

Steps 2-3 can be in any order.

### Actual results

App crashes.

### Expected results

Should change parameter or clamp unusable values.

### Project file & Debug log uploads

[a1.3mf.zip](https://github.com/user-attachments/files/17742722/a1.3mf.zip)
[Log.zip](https://github.com/user-attachments/files/17742729/Log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Crash Log from the OS

```
Extracting libpas PGM metadata failed.

Crashed Thread:        10

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x000000016e914000
Exception Codes:       0x0000000000000002, 0x000000016e914000

Termination Reason:    Namespace SIGNAL, Code 10 Bus error: 10
Terminating Process:   exc handler [6282]

VM Region Info: 0x16e914000 is in 0x16e914000-0x16e918000;  bytes after start: 0  bytes before end: 16383
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      VM_ALLOCATE                 16e910000-16e914000    [   16K] rw-/rwx SM=PRV  
--->  STACK GUARD                 16e914000-16e918000    [   16K] ---/rwx SM=NUL  stack guard for thread 11
      Stack                       16e918000-16e9a0000    [  544K] rw-/rwx SM=PRV  thread 11

Error Formulating Crash Report:
Extracting libpas PGM metadata failed.

Thread 0:: orcaslicer_main Dispatch queue: com.apple.main-thread
0   OrcaSlicer                    	       0x102ead644 Slic3r::GUI::GLModel::Geometry::add_triangle(unsigned int, unsigned int, unsigned int) + 1032
1   OrcaSlicer                    	       0x103320304 Slic3r::GUI::TextLinesModel::init(Eigen::Transform<double, 3, 2, 2> const&, std::__1::vector<Slic3r::ModelVolume*, std::__1::allocator<Slic3r::ModelVolume*>> const&, Slic3r::GUI::Emboss::StyleManager&, unsigned int) + 8324
2   OrcaSlicer                    	       0x102e0441c (anonymous namespace)::init_text_lines(Slic3r::GUI::TextLinesModel&, Slic3r::GUI::Selection const&, Slic3r::GUI::Emboss::StyleManager&, unsigned int) + 576
3   OrcaSlicer                    	       0x102e0a1ec Slic3r::GUI::GLGizmoEmboss::draw_advanced() + 6840
4   OrcaSlicer                    	       0x102e067e8 Slic3r::GUI::GLGizmoEmboss::draw_window() + 400
5   OrcaSlicer                    	       0x102e056f4 Slic3r::GUI::GLGizmoEmboss::on_render_input_window(float, float, float) + 3660
6   OrcaSlicer                    	       0x102dfed00 Slic3r::GUI::GLGizmoBase::render_input_window(float, float, float) + 28
7   OrcaSlicer                    	       0x102df4bf8 Slic3r::GUI::GLGizmosManager::do_render_overlay() const + 880
8   OrcaSlicer                    	       0x102d95b10 Slic3r::GUI::GLCanvas3D::_render_overlays() + 152
9   OrcaSlicer                    	       0x102d925cc Slic3r::GUI::GLCanvas3D::render(bool) + 1368
10  OrcaSlicer                    	       0x102d9f0dc Slic3r::GUI::GLCanvas3D::on_mouse(wxMouseEvent&) + 360
11  OrcaSlicer                    	       0x1053a43e0 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 224
12  OrcaSlicer                    	       0x1053a40dc wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
13  OrcaSlicer                    	       0x1053a3f00 wxEvtHandler::ProcessEvent(wxEvent&) + 56
14  OrcaSlicer                    	       0x1053a4580 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 24
15  OrcaSlicer                    	       0x1052fd084 wxWidgetCocoaImpl::DoHandleMouseEvent(NSEvent*) + 92
16  OrcaSlicer                    	       0x1052f85a8 wxWidgetCocoaImpl::mouseEvent(NSEvent*, NSView*, void*) + 176
17  AppKit                        	       0x186bdd4fc -[NSWindow _handleMouseDraggedEvent:] + 52
18  AppKit                        	       0x18624f558 -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 372
19  AppKit                        	       0x18624f210 -[NSWindow(NSEventRouting) sendEvent:] + 284
20  OrcaSlicer                    	       0x1052e8b40 -[wxNSWindow sendEvent:] + 116
21  AppKit                        	       0x1869184f0 -[NSApplication(NSEventRouting) sendEvent:] + 1604
22  OrcaSlicer                    	       0x10526c414 -[wxNSApplication sendEvent:] + 84
23  AppKit                        	       0x186565d9c -[NSApplication _handleEvent:] + 60
24  AppKit                        	       0x186116020 -[NSApplication run] + 512
25  OrcaSlicer                    	       0x1052e11c4 wxGUIEventLoop::OSXDoRun() + 112
26  OrcaSlicer                    	       0x1053acd14 wxCFEventLoop::DoRun() + 40
27  OrcaSlicer                    	       0x105335478 wxEventLoopBase::Run() + 132
28  OrcaSlicer                    	       0x105311a08 wxAppConsoleBase::MainLoop() + 120
29  OrcaSlicer                    	       0x1052a5514 wxApp::OnRun() + 36
30  OrcaSlicer                    	       0x10535d3f4 wxEntry(int&, wchar_t**) + 164
31  OrcaSlicer                    	       0x102f452b8 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 456
32  OrcaSlicer                    	       0x1020a5444 Slic3r::CLI::run(int, char**) + 14040
33  OrcaSlicer                    	       0x102103a0c main + 172
34  dyld                          	       0x18245b154 start + 2476
```

petercockroach, 2025-01-27:
I'm hitting this constantly on macOS as well

petercockroach, 2025-01-27:
[OrcaSlicer-2025-01-27-133955.txt](https://github.com/user-attachments/files/18564831/OrcaSlicer-2025-01-27-133955.txt)

closed 2025-05-05 by github-actions[bot]