OrcaSlicer/OrcaSlicer #7484: Installer is revoked by windows smartscreen

Culain, 2024-11-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

unrelated

### How to reproduce

1. download from official github repo
2. execute installer

### Actual results

[image]
Translation:
The computer was protected by Windows
Microsoft Defender SmartScreen has prevented the start of an unknown app. Running this app may pose a risk to the PC.
App:
OrcaSlicer_Windows_Installer_V2.2.0.exe
Publisher: Unknown publisher
[Run anyway]  [Do not run]

### Expected results

Installer running without this Message

### Project file & Debug log uploads

Link that was used to download the Installer:

https://github.com/SoftFever/OrcaSlicer/releases/download/v2.2.0/OrcaSlicer_Windows_Installer_V2.2.0.exe

### Checklist of files to include

- [ ] Log file
- [ ] Project file

karrenbauer, 2024-11-14:
i got the same screen....

igiannakas, 2024-11-15:
There's nothing the project here can do - any time a new release is published, Windows is treating it cautiously until enough people have clicked that it's OK, basically building reputation with Microsoft, in simple terms.

What you can do is accept and install it. It shouldn't prevent you from setting up the software.

kylek29, 2024-11-15:
^^ .. this happens with all small scale (relatively speaking) software. For a lot of the utilities I run, I'd be surprised if they didn't get flagged.

Along with igiannakas said, I believe this also has to do with code-signing. If I recall reading correctly in the past, there's newer certificates you can use but Microsoft has made it prohibitive to use them (e.g. fees or something, basically a rich company bypass).

I do think it's counterproductive because it flags so many pieces of software now it's effectively numbed me to the alert in the first place.

closed 2025-02-21 by github-actions[bot]