OrcaSlicer/OrcaSlicer #7489: Reload Preview after Post-processor script to preview updated G-code

ansonl, 2024-11-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The Preview is currently not updated to reflect the latest exported G-code when a Post-processor Script is run in OrcaSlicer after the main slicer runs. Since the new post-processed G-code is the G-code that is exported, the Preview should reload to be consistent and save the user the time of opening a second blank OrcaSlicer window to Preview the exported G-code.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The Preview should reload the rendered G-code after Post-processing Scripts run to show the latest code and save the user the time of opening a second blank OrcaSlicer window to Preview the exported G-code.

### Describe alternatives you've considered

1. Click export gcode file
2. Opening a second blank OrcaSlicer window to Preview the exported G-code.
3. Repeat everytime

vgdh, 2024-11-15:
Some scripts tool long time to post process so it is not very convenient to do it every time you slice the model.
May be some survey needed to make clear what the end user needed.

closed 2025-02-21 by github-actions[bot]

ShaneDelmore, 2025-03-05:
> Some scripts take a long time to post process so it is not very convenient to do it every time you slice the model. May be some survey needed to make clear what the end user needed.

What about loading the preview immediately, then an option to reload it after the script finishes?  There have been some interesting scripts put out lately, from non-planar top layers, to thinner outer layers or brick layers, I think when users are experimenting with things like this it is useful to see the modified gcode.  Currently people I know experimenting in this area go through the hassle of saving the post processed file, then open it in prusa gcode viewer.

eternal-sorrow, 2025-03-26:
+ 1 from me. And for goodness' sake, disable the bot. Automated issue closing shouldn't be a thing.

Mrroot3r, 2025-04-21:
Pretty much the issue I was looking for, these post processing scripts are amazing. its a huge hassle having to have a second copy of orca open to then export gcode. and need to drag it in every time I change a slicer setting. it makes fine tweaking of these scripts take way longer than it should. I will say I have a 5900x and running the bricklayers script with pypy takes like a second for most prints. I can definitely wait a few more for it to refresh the gcode preview every time.