OrcaSlicer/OrcaSlicer #7493: Printer access code isn't saved when leaving Orca Slicer open

Takao-7, 2024-11-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

15.0.1 (24A348)

### Additional system information

- Mac Book Pro 14" 2021 (M1)
- The printer is connected through LAN-Mode

### Printer

Bambu Lab A1

### How to reproduce

1. Slice any object
2. Click "Print Plate", select the printer and print it
3. After you have been redirected to the Device Tab, go back to the Prepare Tab
4. Close the Macbook. Leave Orca Slicer open.
5. Wait until the print is done
6. Open the Macbook
7. Click "Print Plate"

### Actual results

- The printer isn't selected in the drop down menu
- When you select it, you are asked to re-enter the access key
- You might also not find the printer in the drop down menu and you have to press "refresh"

Note that if you close Orca Slicer after starting the print and then re-open it after the print is done, the issue doesn't happen

### Expected results

The access key of a printer is saved

### Project file & Debug log uploads

[Archiv.zip](https://github.com/user-attachments/files/17780403/Archiv.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

This issue also happens in Bambu Studio.

Somewhat similar issues:
- https://github.com/SoftFever/OrcaSlicer/issues/6169
- https://github.com/SoftFever/OrcaSlicer/issues/7439

Berkays, 2024-11-15:
I'm using the workaround in the post you linked. It solves the issue. https://github.com/SoftFever/OrcaSlicer/issues/6169#issuecomment-2330205871

Takao-7, 2024-11-18:
The workaround solves the issue. 
However it’s still a workaround and not a solution.

tflucke, 2024-12-08:
Also seeing this issue.  The linked workaround fixes it for me too.

closed 2025-03-16 by github-actions[bot]