OrcaSlicer/OrcaSlicer #7494: Brim breaks with elephant foot compensation

Vondanta, 2024-11-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Mint

### Printer

Any

### How to reproduce

Check attached image.
This issue has been around for 3 years dating back to prusa slicer
[image]

### Actual results

Using elephant foot compensation results in a gap between the brim and model

### Expected results

^

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

vgdh, 2024-11-16:
that is definitely not right

Jarery, 2024-12-22:
was printing a stealth burner face plate and had same issue, the elephant foot compensation amount is not taken into account with brim

kylek29, 2025-01-21:
Bumped into this the other day, it does cause warping issues because of the disconnection. I had to troubleshoot and came to the same conclusion, elephant isn't being accounted for in the brim calculation.

Related:
https://github.com/SoftFever/OrcaSlicer/issues/3720

BlackRiverCoder, 2025-03-17:
Bug still persist in the newest version 2.3.0-rc. They should fix it.

themechman, 2025-03-25:
Confirming that this is an issue with 2.2. More so, this gap is MORE noticeable on a cool plate (Low temp)

moth4017, 2025-04-08:
still issue in  2.3.0 build 70931e5

luksto, 2025-06-27:
+1 makes Brim useless with elephant foot compensation.

luksto, 2025-09-26:
I dont See it solved in 2.3.1, does anyone else dont have this issue?