OrcaSlicer/OrcaSlicer #7499: Hexagonal slicing with variable z-offset for layer lines

ARDev1161, 2024-11-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

According to this video https://www.youtube.com/watch?v=9IdNA_hWiyE the patent is not valid. It seems to me that this can be easily implemented in a slicer, which will allow for much higher mechanical strength of parts

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

1) the first layer is made by alternating lines of different heights. It is also necessary to be able to choose these two values
2) subsequent layers are made of lines of standard height, but each line will have its own z-offset to touch the previous layer

MxBrnr, 2024-11-16:
A YouTube video cannot be trusted for legal advice.

nubnubbud, 2025-01-22:
`A YouTube video cannot be trusted for legal advice. I would not implement this until the legal situation is fully resolved.`

**TL;DR: While a youtube video should not be trusted for legal advice, you don't need a lawyer for basic knowledge like this. In the US it's common knowledge patents expire after 20 years, and that companies often produce rules and clauses that illegally restrict the public's right, under the assumption that no one will question intimidating documents. Therefore as a US citizen I would boldly ignore this. That patent even has incorrect references, dates and numbers, then uses the incorrect references to claim something new, while simultaneously claiming not 1, but 4 previous patents have addressed their "novel" approach.
It's like a spam email.**

---------------------------------------------------------------------------------------------------------------

**very long but I read it anyway;**
I did not trust the youtube video, so I looked it up personally, and I found nothing incorrect. here is the link, and all the sites seem accurate.

https://www.patreon.com/posts/115566868
https://patents.google.com/patent/US5653925A/en
https://patents.google.com/patent/US20210154911A1

I investigated the patents, and they are real. and the second one absolutely holds no water. not only does it erroneously reference unrelated patents and unrelated images/diagrams, in the cases where it is a typo, it does so attempting to reference already expired patents for the same technique. the length of a patent is common knowledge (like knowing breaking and entering is illegal- you should not need to wait for external or lawyer to assume it is, and if you're wrong, you will be forgiven in thinking so when you find someone else in your house by a broken window) 

**things like patent expiry, and that companies will ignore the law if able or publish unenforceable rules and documentation, are common knowledge and easily found (for example, Nintendo's patent on "moving platforms in video games"). As a citizen, I would ignore this patent, just as I would ignore a man saying I can't ask for water at a restaurant.**

...

HOWEVER...
even if you were to assume the patent is valid... it states the only differences are, in essence; 
"our only difference is that we have partial height layers every other extrusion on first and last layers"
and
"though it is the same technique as in another paper, we are reducing porosity for specifically strength." ... "As noted , a two - stage nozzle or changes in the nozzle velocity and / or changes in the flow rate can be used to create the bead profiles contemplated for use herein"

**but these claims are already made in the original paper.**
for the first claim about partial height layers is handled in several parts of the paper. though it does not mention partial height, it does specifically mention the use of layers with an adjusted extrusion to enable uniform porosity, which in patent language includes doing so with other methods as stated by:

1995 patent:
> As shown in the figures, bead elements may be deposited in successive horizontal layers so that the beads are in such
> horizontal and vertical juxtaposition as to be arranged in a rectangular or an hexagonal array. It is understood that beads may be deposited in a large variety of patterns not shown here. 
> depositing the material in multiple, adjacent elements, with the elements positioned to provide fluid pockets therebetween; and adjusting the rate of dispensing of the material to provide a predetermined porosity in the article thus formed.
> claim 1. 
> A method of making a three-dimensional article by the deposition of solidifiable material onto a receiving surface, with the article having a predetermined porosity comprising that fractional portion of the article that is devoid of such material, comprising the steps of:
> dispensing solidifiable material onto a receiving surface from a dispenser moved in a predetermined pattern to create a three-dimensional article of predetermined size and shape, with dispensing of the material being controlled to sequentially build up the material to form the desired article; 
> claim 16. 
> The method of claim 1 wherein: the rate of dispensing is adjusted to provide a porosity of between 1% and 2%.

but the new paper EXPLICITLY STATES that one of the methods for making those smaller extrusions is by adjusting the "flow rate" or in other words "the rate of dispensing"

and as for the second claim from the new paper, suggesting that decreasing porosity increases strength;

> "These fluid pockets or voids should be contained in the nominally solid portions of the parts. While such parts are less strong and dense than non-porous parts, they are more dimensionally accurate,"...

**THE ORIGINAL PAPER FROM 1995 NOTES THE EXACT EFFECTS AND NECESSITIES THE NEW ONE SAYS ARE NOVEL.**