OrcaSlicer/OrcaSlicer #7510: Orca doesn't detect overhangs facing inwards in vase mode

czlowiekwidmoo, 2024-11-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

8

### Printer

all

### How to reproduce

Same object sliced upside down shows where the overhangs are
[image]
[image]

### Actual results

"Slow down for overhangs" and "fan speed for overhangs" works only on the lower part where overhangs are facing outwards.

### Expected results

In vase mode overhangs should be calculated based only on previous layer overlap, I suppose

### Project file & Debug log uploads

[lampa.zip](https://github.com/user-attachments/files/17792968/lampa.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

henrivdr, 2024-11-18:
Thanks for reporting.  I have the same issue, with the vase mode Bear and actually on normal prints with internal overhangs.

henrivdr, 2025-01-19:
@igiannakas are you able to have as look at this one please?
https://github.com/SoftFever/OrcaSlicer/pull/8080 improves these external overhangs, but in this case somehow internal ones are not detected at all. Referring to the top of the legs.

Shown by the cones as well.  It can be replicated by adding the builtin Cone primitive, then reducing Z height and flipping it.

Not sure if this is related https://github.com/SoftFever/OrcaSlicer/issues/3858

[image]

igiannakas, 2025-01-19:
Internal overhangs are meant to be supported by infill and internal bridging layers. There is no need to slowdown these surfaces. Also the ensure vertical shell thickness function adds additional reinforcement internally for these areas helping them be supported from the inside.

verkkomuikku, 2025-02-01:
Hi igiannakas, contrary to your view I think internal overhangs over sparse infill treated just like outer perimeter overhangs would be great!

On a bigger objects and less sparse infill it would save time and filament greatly, in comparison to the solid infill solution that is currently trying to address this issue. 

It's huge issue actually with particular shaped prints, I've been wasting kg:s of filament trying to come up with slicer settings that would result in fine surface quality but no avail yet. This is really painful experience, it's software issue when the printer is perfectly capable of the quality but anything printed over sparse infill results crappy surface, or the solid infil is just huge waste of material. I just can't make the paths with slicer. I need to start looking at the code. Any tips where to start if I wanted to look into implementing this?

henrivdr, 2025-02-01:
Strangely enough, slowdown is applied on the dome inside the Orca Cube.
But I guess it's because it's effectively an external overhang with a 360° sweep.

As soon as I get to my PC I will send a screenshot of all the geometries I've been comparing and their slowdown, or lack thereof.
TLDR; Its only vase mode and enclosed objects with little to no infill which are affected.

verkkomuikku, 2025-02-01:
I have to note that the partial solid infill and internal bridge is great when there is a flat portion over a sparse infill, like the flat rim on the outside of the model here, you see the internal bridge layer is there and it's working great.

[image]

Here is rest of the top of the model, the surface inside the "vase" curves and gets bad quality when printed, unless I make it ~solid infill.

[image]

Example of Ensure wall thickness set to "All", which makes huge waste of time and material, and really doesn't even work as expected since some layers are not applied with the internal bridge, and solid infill strips print on top of nothing basically, and ruined surface it is. Solid infill is about the only option currently to make nice surface on the curved top portion, and even that is problematic for some reason, any blob of material or even stringing anywhere and it seems to make some anomalies on the surface. It's just really frustrating to deal with this.

[image]

I'm currently printing this Ensure wall thickness set to "No" and using a modifier which adds few extra wall loops and changes to outer/inner order trying to make it nicer, let's see how it goes. 

On previous item I had to find suitable slicer settings to get the walls into ~ballpark what I thought would work and then make custom post processing Python script to slowdown certain layers, and it still ruined about 50% of the time :D Now I must dig into the source try to implement the internal overhang feature as I see it would solve this because external overhangs seem to be working 100% of the time so why not use them here for internal overhangs as well