OrcaSlicer/OrcaSlicer #7516: Allow expansion of interface layers with support expansion

morganvenable, 2024-11-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Support material interface layers often require a bit of expansion to be able to peel them off cleanly in a block.
Shown here with 2mm expansion, the interface layers are not expanded, making it very difficult to peel off in a clean block.
[image]
Interface layers should be expanded with support materials to encourage easy removal.
Suggest a checkbox feature to keep interface layers the same size as the expanded supports.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I'd suggest a checkbox feature to keep interface layers the same size as the expanded supports.
This would make it easy to use, while not creating messy corner cases with people setting different numbers for support and interface expansion sizes.

### Describe alternatives you've considered

I currently have to model supports myself or generate throwaway geometry to encourage the slicer to produce the necessary supports.  It's very time consuming and often leads to print failures if the throwaway layers happen to come unstuck during printing on the support material...
[image]

vovodroid, 2024-11-19:
I guess somewhere in `generate_interface_layers` function.

closed 2025-02-25 by github-actions[bot]

mrnoisytiger, 2025-11-18:
Any chance we can revive this one? Would be helpful for support removal as well, for something to grip onto.