OrcaSlicer/OrcaSlicer #7517: Tree supports should calculate for all models, not per model.

bjaurelio, 2024-11-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am trying to simplify supports by having the same trees support two different models that are mirror images of each other. However, no matter which tree option I try, the supports are always mirrored as well without a single tree in the center supporting both models, which should happen based on parameters.
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Allow a single tree to provide supports to more than one model.

### Describe alternatives you've considered

Print with wasted support material

### Additional context

Allowing a single tree to support more than one model will reduce waste on supports.

MxBrnr, 2025-02-17:
Bump to keep the issue open. I think this is the next major issue that needs to be resolved.

bjaurelio, 2025-02-17:
I am fine with closing the issue. The way to get the same trees to support
multiple models is to combine them into an assembly. The current method may
need better advertising, but it is necessary for the exclude object feature
to function.

MxBrnr, 2025-02-18:
I agree that keeping the supports separate per model is important for, as you said, the `Exclude objects` feature. The issue I see is that Orca does not accurately account for the position of all supports when placing them, but rather ignores them, which causes interfering overlap in the G-code paths.

bjaurelio, 2025-02-18:
That's different from what I was initially opening, but I do see the value
in your issue.