OrcaSlicer/OrcaSlicer #7530: New Nozzle for existing printer not working

MacBoyPro98, 2024-11-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

macOS Sequoia 15.1.1 (24B91)

### Additional system information

CPU: M2 Max
Memory: 32 GB

### Printer

Any

### How to reproduce

1. Go to the printer selection dropdown
2. Scroll down to the bottom and select `------ Create Printer ------`
[image]

3. In `Create Type` field, select the `Create Nozzle for Existing Printer` option
[image]

4. Try selecting a Printer from the `Select Printer` dropdown in the `Printer` field

### Actual results

The dropdown does not display a list of currently available printers:

https://github.com/user-attachments/assets/6a9ecdb7-1974-4217-9cfd-04383b6f38e1

### Expected results

The dropdown should be populated with a list of currently available printer profiles to select from.

### Project file & Debug log uploads

[debug_Wed_Nov_20_09_16_02_5548.log.0.zip](https://github.com/user-attachments/files/17833823/debug_Wed_Nov_20_09_16_02_5548.log.0.zip)
[nozzle_test.3mf.zip](https://github.com/user-attachments/files/17833847/nozzle_test.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Twikzer, 2024-11-22:
same problem here, but on windows 11, with nvidia rtx 4000 gpu, and ryxen 7 5000 cpu

living-relation, 2024-12-01:
same issue. drop dn menu does not do anything.

Leo-Bourbon, 2024-12-19:
Same issue here. Using 2.2.0 on Windows 11, I can give more info on my specs if needed.

closed 2025-03-27 by github-actions[bot]