OrcaSlicer/OrcaSlicer #7536: Missing documentation on supports

Beep6581, 2024-11-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Manjaro

### Printer

n/a

### How to reproduce

Hover over "Top/Bottom Z distance" in Support > Advanced. The tooltip points to this URL, but such a page does not exist:
https://github.com/SoftFever/OrcaSlicer/wiki/support

### Actual results

URL does not exist.

### Expected results

Existing documentation regarding supports.

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

closed 2025-02-27 by github-actions[bot]

Beep6581, 2025-03-24:
Curb your bot. Issue still valid.

ArsenicBismuth, 2025-12-17:
Yeah this happens to multiple links, all redirect to an old links.