OrcaSlicer/OrcaSlicer #7538: Slicing produces broken GCODE

Ohmarinus, 2024-11-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10 22H2

### Additional system information

AMD Ryzen 7 4800H with Radeon Graphics
Nvidia GeForce RTX 2060
32Gb

### Printer

Bambu Lab X1C

### How to reproduce

Slice, and see this result:
[image]

I get this result with the specific settings I've used. But if I change settings around I get a result that is not broken. The mesh reports as ok.

### Actual results

See 'How to reproduce', the issue is clear.

### Expected results

The expected result is no error in the generated GCODE.

### Project file & Debug log uploads

Project file:
[roci_error.zip](https://github.com/user-attachments/files/17848054/roci_error.zip)

Debug:
[debug_Thu_Nov_21_16_11_12_35504.log.zip](https://github.com/user-attachments/files/17848106/debug_Thu_Nov_21_16_11_12_35504.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

discip, 2024-11-21:
@Ohmarinus
You forgot the **`%`**.

[image]

Ohmarinus, 2024-11-21:
> @Ohmarinus You forgot the **`%`**. [...]

Thanks, that was it.. Now I feel so dumb haha. Okay, that was indeed the issue, and not a bug in Orca. Thanks a lot!

closed 2024-11-21 by Ohmarinus

discip, 2024-11-21:
Nevermind.
You are welcome!