OrcaSlicer/OrcaSlicer #7541: Conical slicing implementation

thisnyjared, 2024-11-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Supports are messy, wasteful and just plain annoying. Conical slicing-though limited by the free angle of your printer-is one that can reduce support use. This idea is one similar to many Full control xyz ideas though not affiliated. It entices the printer to print at angles not just flat, and may benefit the 3d printing community.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

This is an idea that has existed for a while now and nobody seems to want to implement it. I would but am not skilled enough yet :) I wish to bring this idea to the attention of other people in hopes that someone will implement this feature-or at least start-for the good of the community. Examples of conical slicing and directions on how to do it manually are as follows: https://www.youtube.com/watch?v=1i-1TEdByZY, https://www.printables.com/model/315702-supportless-pipe-conical-slicing, https://www.cnckitchen.com/blog/guide-how-to-use-conical-slicing.

### Describe alternatives you've considered

1. Improving supports
2. increasing speed of supports
3. Using temporary and reusable supports (sometimes used at print farms)

### Additional context

Thank you for taking the time to read this.
[image]

kylek29, 2024-11-22:
This is one of those ideas that seemingly got a lot of traction and then never got implemented. A lot of research was done around non-planar printing as a way to improve surface finishes on rounded objects.

I suspect the patent might be preventing people from implementing it: https://patents.google.com/patent/US10005126B2/en

thisnyjared, 2024-11-23:
I see. Well at the time of writing I did not know of arc overhangs another solution, however is not limited by the free angle. https://www.cnckitchen.com/blog/arc-overhangs-in-prusaslicer this however I am afraid that this dose not have the research as conical slicing dose so it may be the better option in the long run.

nubnubbud, 2025-01-22:
> I see. Well at the time of writing I did not know of arc overhangs another solution, however is not limited by the free angle. https://www.cnckitchen.com/blog/arc-overhangs-in-prusaslicer this [...]

I would be interested in seeing this make arc overhangs. they can be a little fragile and droop, but the slight angle from this should help strength, rigidity, and adhesion, I think.