OrcaSlicer/OrcaSlicer #7543: Dragging the Orca window causes delays and severe lag

hechuan949494, 2024-11-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10 

### Additional system information

AMD Ryzen 9 5950X

### Printer

Bambulab X1C

### How to reproduce

Just drag the windows
[image]

### Actual results

The window moves very slowly and cannot keep up with the speed of the mouse movement

### Expected results

Drag the windows smoothly

### Project file & Debug log uploads

None

### Checklist of files to include

- [ ] Log file
- [ ] Project file

frank-pian, 2024-11-22:
I have the same problem, and it's really bothering me. Prusa slicer is ok.

Ergonomicmike, 2024-11-22:
Interesting. I tried it.

No problem on my system. (Win10, dual monitors, Intel internal GPU driving one monitor, NVIDIA GPU driving the other. (1920x1080.)

My guess is that slow redraw behavior is a video driver issue. Or maybe the Graphics Setting in Win10?

frank-pian, 2024-11-23:
> My guess is that slow redraw behavior is a video driver issue. Or maybe the Graphics Setting in Win10?

It's very slow on my Windows 10 system with an NVIDIA 1060s GPU, but there's no issue on another PC with Windows 11 and a 4060 GPU.

Ergonomicmike, 2024-11-23:
I have a GTX 1650 Super, which is similar to your 1060 and I have no issue here with dragging the window.

I'm running 565.90 Studio version. I have Orca Slicer set to "High Performance" in Windows Graphics setting, FWIW.

MS Edge WebView2 Runtime 98.0.1108,43. I don't keep my Visual C++ Redistributables up to date. But maybe?

Someone here was having a display problem with Orca that was caused by using a non-vanilla Windows Theme.

ZAKsPop, 2024-11-25:
I have this issue too. It also happens in QidiStudio. Both are based off of Bambu Slicer. I would guess the issue is in the source code of Bambu Slicer.

ZAKsPop, 2024-11-26:
> #3111 sounds like a duplicate of this. Can you guys confirm your mouse's polling rate?

My polling rate was at 500. I had to slow it down to 125 to make the issue stop. As I am not a gamer I don't see any difference other than the issue being gone but I'm sure this isn't an option for some people.

jbanales1, 2025-02-24:
Just installed OrcaSlicer and I'm experiencing the same issue. Changing the polling rate from default (1000) to 125 seems to fix the issue but I would rather not have to do that.

closed 2025-06-01 by github-actions[bot]